rodizio1/EZ-WifiBroadcast #72: wifibroadcast 1.6 rc3 not support RTL8812AU for wifi acces point

roque-canales, 2017-12-30:
Hello I just upgraded wifibroadcast,and my setup with rtl8812au en receiver side for wifi acces point don't work anymore.

I return back to 1.6 rc2 and it work perfectly.

rodizio1, 2017-12-31:
Strange, I haven't (knowingly atleast) changed anything in the kernal or elsewhere regarding that. Do you get any error messages?

Can you please set DEBUG=Y in wifibroadcast-1.txt, start up the RX Pi with the Realtek dongle attached, wait a minute and then insert a USB memory stick? You will find a file called debug.txt on it, please post that here so I can see what's going on.

rodizio1, 2018-01-01:
Duh. Sorry, it's a stupid bug. Just found that I forgot to re-compile the (external) RTL8812 driver after changing the kernel in 1.6RC3.

Will be fixed in 1.6RC4 soon.

roque-canales, 2018-01-01:
Thank you for this feedback!

roque-canales, 2018-01-01:
Thank you for this feedback and happy new year!

geofrancis, 2018-01-01:
do you have a ETA for rc4?

vaidyasp, 2018-01-10:
@rodizio1 Also Mediatek MT7601 is not working in 1.6. It works fine with 1.5. Just changing the sd card with 1.5 firmware on same hardware does the job.
Try to solve that in next release.

vaidyasp, 2018-01-10:
Screenshot and errorlog.txt both files attached. Feel free to let me know if any other data is required or testing is to be done.

[image]

[errorlog.txt](https://github.com/bortek/EZ-WifiBroadcast/files/1618430/errorlog.txt)

rodizio1, 2018-01-10:
thanks a lot. This is strange, the card is detected by the kernel, but just the rt2800usb kernel module doesn't get loaded, that's why it doesnt work. Question is still, why the driver module doesnt get loaded. Could you please make another errorlog.txt from version1.5 to compare?
Just set a wrong frequency like 3000 or so to make it throw an error and write the log.

vaidyasp, 2018-01-12:
Yes its strange. Its detected but says not detected. Please check attachment for error on v1.5 with frequency changed to 3000.

[image]
[errorlog.txt](https://github.com/bortek/EZ-WifiBroadcast/files/1625312/errorlog.txt)

rodizio1, 2018-01-14:
Well, this is interesting. All the time I thought the MT7601 was also handled by the rt2800usb chipset (since it also has a Ralink USB ID). However, it's actually handled by the mt7601u driver, which is not compiled-in with the kernels in 1.6. 

I'll re-enable it in the next release. Be aware though that without the txpower settings (which are only in the rt2800usb driver) you cannot increase txpower, so it will probably be around only 10-15mW. Not sure if it's possible to increase transmit power in the mt7601u driver.

rodizio1, 2018-02-11:
Fixed in 1.6RC4.

closed 2018-02-11 by rodizio1