rodizio1/EZ-WifiBroadcast #74: EZ-WifiBroadcast and SPI devices

andrewhazi, 2018-01-11:
Hi Rodizio,

First off, I wanted to thank you for all the work you've put in to this, it's really made it very easy to get up and running with wifibroadcast. I have run in to an issue using SPI devices though:

Up until version 1.3, when I enabled SPI using raspi-config, /dev/spidev0.0 and /dev/spidev0.1 appeared as devices. From version 1.4 however this has stopped working. I've tried various things such as manually editing config.txt but can't get SPI working from version 1.4 onwards. My next step is to do a detailed comparison of the v1.3 and v1.4 images to inspect for differences (I notice that the raspberry firmware was updated in v1.4) however I'm wondering if you might have a simple answer/fix as to why SPI isn't working from v1.4 onwards.

Thanks

rodizio1, 2018-01-11:
I removed a lot of drivers and modules in the kernel in 1.4 to make it smaller and speed-up compile time, SPI also got removed there. Wasn't aware that people want to use this until lately. In 1.6, I have enabled it  in the kernel, but maybe something is still missing?

andrewhazi, 2018-01-14:
Thanks for the info, I'll try and work with this to see if I can get it working.

rodizio1, 2018-01-18:
I've tried it now, it was indeed CONFIG_SPI_SPIDEV missing. Will be added in the next release.

tictag, 2018-01-20:
Haven't yet got to the part where I'm using SPI but my thermal camera module will eventually use it. Am I correct to assume that SPI is not functional in 1.6RC3?

andrewhazi, 2018-01-20:
Thankyou, any rough idea of the next release date? Tictag I haven't been able to use SPI devices in 1.6RC3 due to the issues mentioned above.

tictag, 2018-01-21:
You're right, after booting there is no /dev/spidev0.0 listed.

For info, however, after running the following code from [pylepton](https://github.com/groupgets/pylepton), `/dev/spidev0.0` does automagically appear. I have no clue what the code does to make this happen, I'm just letting you know that this may be a workaround to having to re-compile the kernel.

```
from pylepton import Lepton
with Lepton() as l:
  a,_ = l.capture()
```

rodizio1, 2018-02-11:
1.6RC4 is released which includes the spi kernel modules now.

closed 2018-02-11 by rodizio1