rodizio1/EZ-WifiBroadcast #75: Quick Question: autostarting wifibroadcast

tictag, 2018-01-15:
@rodizio1 I just spent the entire of yesterday trying to find out how wifibroadcast auto-started. I went through: rc.local, .bashrc, init.d directory, systemd & cron auto-start systems (_Google was my friend_) and could only find where the "WifiBroadcast v1.4RC4 ....." text is displayed via /init.d/raspi-config. I can clearly see from the bootup sequence behaviour that the "profile" bash script in /wifibroadcast-scripts is how wifibroadcast is executed but just can't find where this script is called from.

QQ: Where is the "profile" bash script called from?

matiaspl, 2018-01-15:
Mind the dot. It's in /root/.profile

tictag, 2018-01-15:
Wow, what a learning curve. So I've now learned that wifibroadcast auto-logs into the console on boot using `systemctrl set-default multi-user-target`, that something called a 'symbolic link' exists between `/etc/systemd/system/autologin@.service`  and `/etc/systemd/system/getty.target.wants/getty@tty1.service`, which is configured to login as `root` and that the `.profile` file is autorun on login to the console, so by executing `tx_rawsock` within `.profile` in `root`'s home directory wifibroadcast automatically runs.

oh, and grep doesn't search through hidden files unless you explicitly tell it to.....grrrr.

Linux sucks.

rodizio1, 2018-01-15:
Yes, there is autologon configured in systemd for tty1-tty12. After logon, the shell (bash in this case), is started which then starts the .profile script. The .profile scripts then looks on what tty it is running to decide what functionality (video rx, osd, telemetry, joystick r/c, etc.) is to be run on that tty.

Yeah, Linux is a mess.

tictag, 2018-01-15:
LOL! Hilarious! Never seen Linux 'blue screen' before ;)

closed 2018-01-27 by rodizio1