rodizio1/EZ-WifiBroadcast #77: Missing telemetry OSD

lgangitano, 2018-01-16:
I've been able to get telemetry OSD on RX just a couple of times but not matter how hard I try it does not work anymore.

Attached a keyboard to TX, on tty2 I see the packet counter increasing so I suppose the TX part is correct (as per the Ardupilot Raspberry instruction I'm using 921600 baud rate on /dev/serial0).

On RX, video is perfect, but telemetry does not show up on OSD. On tty2 I see a shared memory error (filesystem read-only) and rx_stats->lost_packet_cnt increasing. It looks like the issue is on the RX side but I'm stuck.

Can you please provide some directions for debug? I'm also interested in analyzing a network dump but I can't tell telemetry packets from video packets.

rodizio1, 2018-01-18:
EZ-Wifibroadcast supports only up to 115200 baud. Actually 912600 is not "officially" supported on the Pi, I guess the Ardupilot guys use these high baudrates because they want to transmit large very logfiles or something. But for telemetry and mission and parameter up/download 115200 should be fine (still twice as fast as those 433Mhz mavlink telemetry radio modules).

Set both the flightcontrol and the serialport speed in the wifibroadcast-1.txt config file to 57600 or 115200 and it should work. Also make sure you have flow control disabled in your flightcontrol.

rodizio1, 2018-01-28:
Forgot something to add: Depending on configuration of those SR_ ... parameters on the flightcontrol, it can happen that the flightcontrol does not send telemetry messages until it is told so by missionplanner/qgroundcontrol etc. Best is to configure it so that it sends telemetry right away after startup.

Also see the wiki article on which message types the osd uses and recommended update rates: https://github.com/bortek/EZ-WifiBroadcast/wiki/OSD-MAVLink-message-types

lgangitano, 2018-01-28:
Thanks! That was definitely the case. I'm now poking with SR2_ parameters (I've connected the RPi on Telem2) and confirmed telemetry working when QGroundControl is connected.

Once I find a stable configuration that does not need QGC and is kept after a reboot, I'll post it here.

rodizio1, 2018-02-01:
Great, I'll add your findings to the wiki then.

maZZiveAttack, 2018-02-17:
Hello people.
Could anyone help? I'm wanting to use ezWifibroadcast with a Pixhawk Cube. Currently the video is working just fine. The display, with the OSD, is showing what needs to be shown for my liking. But the date/telemetry does nothing.
I know a lot of people are using pixhawk cube, could someone using a cube please lay out what I need to do. It's great having all the settings etc etc but it means nothing to be as I've never programmed before. I've never used the Mavlink protocol before. 
I've made some changes in some of the setting in the GCS. But forget all that because it still doesn't work.
Is DJI or Parrot paying you guys NOT to post anything by way of a step-by-step guide? I find it incredible that this is a game-changer for fpv yet I can't find a single how-to-do video or set of instructions anywhere on youtube or here, or anywhere. Forget the posted setting. They mean nothing to people whom don't understand that language. 
Be helpful for once and stop posting show-off videos of your gear working without explaining how you do it. It's boring!
Be helpful with step-by-step guides, ffs, please

closed 2018-03-30 by rodizio1

ManishSahu53, 2018-04-11:
@lgangitano Can you please tell us how you successfully able to get telemetry data?  I am getting Video data but not getting telemetry. Only Mode status Alt Hold is showing rest all is blank.
Thanks