rodizio1/EZ-WifiBroadcast #79: linux-4.9.28-dwc-otg.patch issues

tictag, 2018-01-22:
All other patch files work perfectly, except `linux-4.9.28-dwc-otg.patch`. When this patch is executed the console reports errors similar to, "Hunk 1 OK, Hunk 2 failed....." (didn't capture the exact error). When subsequently compiling the kernel, the compile fails when these files are reached.

When comparing this patch file to v4.9.35 of the RPi kernel source code, most of the changes appear already to be implemented and the one section that isn't doesn't intuitively look correct - please see attached screenshots for detailed analysis.

As a workaround for the patch failing, I manually updated the affected source files (`dwc_otg_driver.c` & `dwc_otg_hcd_queue.c`), after which the kernel compiled OK.

For others, would you be able to review the attached and update the patch file accordingly?

[image]
[image]
[image]
[image]
[image]

rodizio1, 2018-01-23:
I made these patches for 4.9.28 initially. It's this feature (which turned out to not work later anyway ...): https://github.com/raspberrypi/linux/pull/2067. Shortly after, a new Raspbian release with 4.9.35 came out which had that particular patch already integrated, so with 4.9.35, the patch isn't needed.

I'm not sure what your manual changes do, I'd rather revert them.

tictag, 2018-01-23:
Ahh, got you. Might be worthwhile modifying the readme.md in /kernel to mention this, save other newbies from making the same mistake, especially as it took quite some time to get the place where it failed! ;) Suggest the following change:

FROM: `Patches also apply to kernel 4.9.35 included in Raspbian 2017-07-05`
TO: `All patches except linux-4.9.28-dwc-otg.patch also also apply to kernel 4.9.35 included in Raspbian 2017-07-05`

rodizio1, 2018-01-27:
Well, it's quite clear when looking at the sources. Regarding the image with the text "Patch deletes then reinserts the same code": No it doesn't, there are some "//" added.

I've changed the text in the readme.

closed 2018-01-27 by rodizio1