rodizio1/EZ-WifiBroadcast #81: How to use internal RP3 wifi to join an access point?

neilyoung, 2018-01-23:
Hi,
great work, indeed, thanks for this.

I'm trying to enable the internal Raspi 3 wifi so that it joins an access point aside. Up to now to no avail.

I was trying to configure it in /etc/network/interfaces as well as in /etc/wpa_supplicant/wpa_supplicant.conf. Didn't work up to now. What do I have to do? 

Any help would be appreciated.

rodizio1, 2018-01-23:
I have removed all that network-manager and wpa supplicant stuff etc. because it tends to mess with the network interfaces. I'd configure it manually and make sure it doesn't touch any other network interfaces (i.e. the wifi card used for wbc).

neilyoung, 2018-01-23:
@rodizio1 Thanks, I would do that too.

I managed to make wpa_supplicant run. The simple reason why it didn't work is: It wasn't installed. 

So I installed it and managed to join my network but as soon as this happened, the wbc connection broke, my entire 2.4 GHz networks went down. It seems, that wbc immediately switched over to this new connection once it appears.

Would you mind to be a bit more specific regarding "I configure it manually?". I also tried this via /etc/network/interfaces, but I couldn't make it run either. 

TIA

closed 2018-01-25 by neilyoung