rodizio1/EZ-WifiBroadcast #82: Impossible to calibrate ardupilot sensor through mission planner over UDP

roque-canales, 2018-01-23:


rodizio1, 2018-01-23:
I think this is probably caused by an issue with outdated mavlink libraries that has recently been identified. 

If you have some basic linux knowledge, you can try if replacing the librarirs with a newer version, see here:https://www.rcgroups.com/forums/showpost.php?p=39028277&postcount=3741

If not, this will also be included in 1.6rc4 (which I hope to have ready soon)

roque-canales, 2018-01-24:
thank you for this feedback Rodizio!

rodizio1, 2018-02-11:
1.6RC4 is released, hopefully it should work now.

rodizio1, 2018-03-30:
@roque-canales Did you have a chance to test?

roque-canales, 2018-03-30:
Hello, Yes I test it, there is many improvement, now we can see many other data, so it's perfect !

About calibration I can calibrate the accelerometers but no the compass..... But I think that there is a mission planner mistake.

I will retest it.

rodizio1, 2018-04-01:
Thanks for reporting back.

roque-canales, 2018-05-05:
it works (copter 3.5.5 and WBC 1.6!!

you can close the issue

careyer, 2018-11-01:
This issue will be solved with the upcomming release with MAVlink v2

closed 2018-11-01 by careyer