rodizio1/EZ-WifiBroadcast #83: Under-voltage or over temperature on TX!

neilyoung, 2018-01-23:
Getting this warning on RX screen. I can see, where it is generated, but I don't know the reason. Temperature and voltage seem to be fine...

rodizio1, 2018-01-23:
If you have just started the Pi and get the message it can't be too high temperature (that would take longer to heat up), then it's bad power supply and/or wiring.

neilyoung, 2018-01-23:
@rodizio1 Hmm. I don't think it is temperature related. And the hw design is not mine. But it seems to be pretty robust (it is the Sunfounder robot car https://www.sunfounder.com/robotic-drone/smartcar/smart-video-car-kit.html), but of course it might have issues. 

Is there a way to remove or to suppress it?

geofrancis, 2018-01-24:
have you checked the voltage that the pi is getting is actually 5v?

neilyoung, 2018-01-24:
4.82

geofrancis, 2018-01-24:
thats low. you should get another regulator.

neilyoung, 2018-01-24:
It is this https://www.amazon.de/SunFounder-DC-DC-Converter-Module-Raspberry/dp/B014KSY13O, fed with 12 V input.

geofrancis, 2018-01-24:
you should use a separate bec for the pi.

closed 2018-01-25 by neilyoung

neilyoung, 2018-01-25:
@geofrancis No the same happens to a brand new RPI3 powered with the original power plug or a power pack 5V 2A. Maybe the expectations need to be lowered.

rodizio1, 2018-01-27:
The stepdown regulator on your car is rated 2A. The instructions in the wiki state you need a 3A power supply. The wires are surely not atleast 0.5qmm like it is recommended in the instructions. USB power banks suck in general, the instructions say not to use them. You are measuring 4.82V, the instructions say you need 5V ...

Just follow the wiring instructions and it'll work fine.