rodizio1/EZ-WifiBroadcast #87: What is tx_rawsocket and how it is build?

neilyoung, 2018-01-24:
In .profile there is a tx_rawsocket launched, but I can't find that in the sources. How is it build?

And what is the advantage using tx_rawsocket vs. tx? The web is full of receipts for parameters of tx. Do they apply to tx_rawsocket as well? Is there also an rx_rawsocket?

Questions over questions...

neilyoung, 2018-01-25:
Unfortunately this community is not that responsive as I would like to have it or my questions are too stupid. I'm leaving EZ-WBC now, using GStreamer instead. This gives a bit more delay, but at least things are not that experimental and the pool of information is bigger.

Thanks for all the help and good luck with this project

closed 2018-01-25 by neilyoung

careyer, 2018-01-25:
@neilyoung : Your question about tx_rawsocket was a good one. However you have to consider that "this community" is rather a one-man-show. Rodizio simply can't answer all the questions and moreover it is not possible to do it in no time.

neilyoung, 2018-01-25:
@careyer :) I just wanted to be a bit provokant. It is really so, that I like this solution, the awesome low delay makes it #1 amongst the streaming solutions. But on the other hand, I spent several days into various small problems to no avail, this is too much compared to what can be pulled out of the box with gstreamer (well, not really, but after a while of dealing with it, it seems so to me)

rodizio1, 2018-01-27:
The latest sourcecode is always included in the image. How about taking 5 minutes for actually looking at stuff yourself before using other people's time? Sourcecode on Github will be updated when 1.6 final will be released.

The problems you have reported so far are caused by you not reading the instructions (like the undervoltage warning or the kbitrate display thing).

Your problems with modifying the image to suit your special needs are caused by your lack of linux knowledge.

neilyoung, 2018-01-27:
@rodizio1 Oh yes, thanks. Get lost

matiaspl, 2018-01-28:
Well that escalated quickly... neliyoung, you could spend more time on
thinking how open souce community works and how you can commit your own
free time to make something good even better. Human relations don't work on
a "get lost" basis. And humbleness is a virtue, you know ;)

tictag, 2018-01-29:
@rodizio1 your contributions to this project are acknowledged, respected and appreciated by all, and trolled only by one. Keep up the great work chap. I have blocked and reported user @neilyoung to GitHub's abuse reporting services.

Trolls have no positive role to play on this platform.

bortek, 2018-01-29:
Agree with tictag. I have done this already for 2 days ago when I first read this. He will no loger be able to comment here. Its good that Github have got several reports from different users, hope they will ban him from Github. Such attitude has no place in the public community.