rodizio1/EZ-WifiBroadcast #88: 1.6RC3 no flightmode in OSD

FritzWalter123, 2018-01-25:
Hello,
so far, all informations from mavlink will be displayed on screen except flightmode. 
Not implemented until now or a bug?

rodizio1, 2018-01-26:
Not implemented (yet). 

There are two users who made extensions to the OSD for version 1.5 and also posted the code, but I haven't found time yet to integrate it.

If you want to try it yourself:
https://www.rcgroups.com/forums/showpost.php?p=38130691&postcount=3163
https://www.rcgroups.com/forums/showpost.php?p=38138774&postcount=3172

FritzWalter123, 2018-01-30:
Thanks a lot for the great job here!  

I try it out on my 1.6RC3 image. but get a segmentation fault when starting the osd binary. I think something has changed to 1.5.  I will spend some more time to this the next days.

FritzWalter123, 2018-02-01:
Hi...i merged the osd stuff in src from 1.6rc3.  There were some merge conflicts to 1.5. Seems to work now. Do you want to have the src ? What is the best way to forward it?  Regards

rodizio1, 2018-02-01:
Now that was fast. Did you merge both? Would be great if you can attach it here (or on the rcgroups thread) or upload it somewhere.

FritzWalter123, 2018-02-02:
[osd_patch_for _16RC3.tar.gz](https://github.com/bortek/EZ-WifiBroadcast/files/1688386/osd_patch_for._16RC3.tar.gz)

FritzWalter123, 2018-02-02:
[osd_patch_for _16RC3.tar.gz](https://github.com/bortek/EZ-WifiBroadcast/files/1688487/osd_patch_for._16RC3.tar.gz)
A crosscheck would be a good idea. Only connected on ground with my apm and not flown yet.

rodizio1, 2018-02-02:
Thanks for helping, really appreciate it. I'll look into it on the weekend.

rodizio1, 2018-02-12:
features are integrated in 1.6RC5 now.

closed 2018-02-12 by rodizio1