rodizio1/EZ-WifiBroadcast #89: output for diagnostic leds.

geofrancis, 2018-02-06:
would you be able to add support for outputting basic information via an led by setting a gpio pin high?

something like:

set a pin high if 
USB joystick detected 
wifi cards detected
camera detected
hotspot client connected
telemetry detected
writing to USB
overheating/low voltage

it would make diagnostics just a matter of looking to see what lights are on.

rodizio1, 2018-02-07:
Cpuload, temperature and undervoltage errors on the TX will be shown in the RX OSD starting from 1.6rc4. 

It's possible to add leds, but often wouldn't help, for example it can happen that the cam gets detected by the Pi, but then it doesn't start transmitting because the cable is not seated right. The only way to detect that is to start the actual transmission with raspivid, but then, if you're already transmitting, you don't really need a led anymore.

Also I think having like a lot of error-codes with different colored leds blinking etc. is also not very user-friendly, I would definitely forget the meanings :) Maybe one led with a meaning of 'no errors detected' would make sense on the TX.

For the RX I think this can be better done via some other means, for example by using whatever mavlink message that might allow to send notifications to missionplanner. For the bootup process and hotspot connection process (when missionplanner is not yet connected) I always wanted to look into speech output via the headphone/speaker jack, but haven't got around looking into that, too much other stuff still on the todo list.

ivanskj, 2018-02-07:
I put a RGB-LED on my setup giving me clues to where it was in the process. 
Red was an error of some kind. 
Blinking blue was waiting for camera-input. Solid blue was camera found (Ping IP on QX1). 
Green was succesfull start of TX.
Purple meant uploading data to USB.

All was controlled by a Python script and calls made from .profile. Easy and intuitive Feedback.

But a message to the Ardupilot via MAVLINK would be a way better solution

carcinogoy, 2018-02-08:
Can you go into more detail on the python script and the wiring please?

ivanskj, 2018-02-11:
Wiring and some code here:
https://www.youtube.com/watch?v=b4_R1eX9K6s

I made a ledcontroller.py real fast that waits for input on a pipe:
`import RPi.GPIO as GPIO
import os, sys
from time import sleep

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)

GPIO.setup(18,GPIO.OUT) #RED
GPIO.setup(23,GPIO.OUT) #GREEN
GPIO.setup(24,GPIO.OUT) #BLUE

pipe_path = "/tmp/ledpipe"
if not os.path.exists(pipe_path):
    os.mkfifo(pipe_path)
# Open the fifo. We need to open in non-blocking mode or it will stalls until
# someone opens it for writting
pipe_fd = os.open(pipe_path, os.O_RDONLY | os.O_NONBLOCK)
with os.fdopen(pipe_fd) as pipe:
    mode = ''
    while True:
        message = pipe.read()
        if message:
            mode = message.rstrip()
            #print("Received: '%s'" % message)
        if mode == 'red':
            GPIO.output(24,0) #BLUE
            GPIO.output(23,0) #GREEN
            GPIO.output(18,1) #RED
        if mode == 'blue':
            GPIO.output(24,1) #BLUE
            GPIO.output(23,0) #GREEN
            GPIO.output(18,0) #RED
        if mode == 'green':
            GPIO.output(24,0) #BLUE
            GPIO.output(23,1) #GREEN
            GPIO.output(18,0) #RED
        if (mode == 'redgreen') or (mode == 'greenred'):
            GPIO.output(24,0) #BLUE
            GPIO.output(23,1) #GREEN
            GPIO.output(18,1) #RED
        if (mode == 'bluegreen') or (mode == 'greenblue'):
            GPIO.output(24,1) #BLUE
            GPIO.output(23,1) #GREEN
            GPIO.output(18,0) #RED
        if (mode == 'bluered') or (mode == 'redblue'):
            GPIO.output(24,1) #BLUE
            GPIO.output(23,0) #GREEN
            GPIO.output(18,1) #RED
        if mode == 'all':
            GPIO.output(24,1) #BLUE
            GPIO.output(23,1) #GREEN
            GPIO.output(18,1) #RED
        if mode == 'off':
            GPIO.output(24,0) #BLUE
            GPIO.output(23,0) #GREEN
            GPIO.output(18,0) #RED
        if mode == 'blinkblue':
            while True:
                GPIO.output(24,1) #BLUE
                GPIO.output(23,0) #GREEN
                GPIO.output(18,0) #RED
                sleep(0.2)
                GPIO.output(24,0) #BLUE
                sleep(0.2)
                message = pipe.read()
                if message:
                    mode = message.rstrip()
                    break
        if mode == 'blinkred':
            while True:
                GPIO.output(24,0) #BLUE
                GPIO.output(23,0) #GREEN
                GPIO.output(18,1) #RED
                sleep(0.4)
                GPIO.output(18,0) #RED
                sleep(0.4)
                message = pipe.read()
                if message:
                    mode = message.rstrip()
                    break
        if mode == 'cycle':
            while True:
                GPIO.output(24,1) #BLUE
                GPIO.output(23,0) #GREEN
                GPIO.output(18,0) #RED
                sleep(0.2)
                GPIO.output(24,0) #BLUE
                GPIO.output(23,1) #GREEN
                GPIO.output(18,0) #RED
                sleep(0.2)
                GPIO.output(24,0) #BLUE
                GPIO.output(23,0) #GREEN
                GPIO.output(18,1) #RED
                sleep(0.2)
                message = pipe.read()
                if message:
                    mode = message.rstrip()
                    break
        sleep(1)
        if mode == 'quit':
            GPIO.cleanup()
            sys.exit(0)`

start the ledcontroller python file somewhere in .profile and continue:
`python /home/pi/led/ledcontroller.py &`

And calls to change color from the .profile could look something like this:
`echo blinkred > /tmp/ledpipe
sleep 3`

rodizio1, 2018-04-01:
I have added your script here so it doesn't get lost:
https://github.com/bortek/EZ-WifiBroadcast/tree/master/usercontributions

closed 2018-04-01 by rodizio1

geofrancis, 2018-04-02:
very cool i am going to try this out.