rodizio1/EZ-WifiBroadcast #90: connect to usb serial devices ground pi via hotspot.

geofrancis, 2018-02-07:
I have a RTK gps and a serial servo controller that I connect to over usb using mission planner, if you could forward usb serial devices to a udp port then I could run it all over the hotspot.

rodizio1, 2018-02-11:
sorry I don't have any devices to test that with and too many other things on my list.

closed 2018-02-11 by rodizio1

geofrancis, 2018-02-25:
I have done this before when i used to hack routers, I think it was ser2net i used to forward a serial port so i could monitor the battery management controller over wifi , it just forwards a serial port to a network port, but that was about 10 years ago when i was running openWRT I am not sure how i would add it to wifibroadcast or if it would interact with wifibroadcasts usage of serial ports.