rodizio1/EZ-WifiBroadcast #91: Add video switch for chose in flight between raspi cam and usb webcam

roque-canales, 2018-02-07:


careyer, 2018-02-07:
+1 vote for such a feature. It would be great to have a feature to switch between two different video sources in mid-flight. I want to use something similar to switch between HD-Video and Thermal-Video. 

@rodizio1 : You might be able to adapt some parts of Franks (dinos) OSD for that. He implemented a function to switch between different OSD screens on the GroundPi via the USB/Taranis. As far as he explained to me he writes all the channels from the Taranis to a memory structure in the GroundPi and from there all channels can be read and used for various things, e.g. to switch OSD screens 1...3 by a 3 position switch.

That would be AWESOME!

rodizio1, 2018-02-11:
sorry, this is way too much work at the moment and also probably wouldn't work as there are different usb cameras I cannot test with.

closed 2018-02-11 by rodizio1