rodizio1/EZ-WifiBroadcast #94: Using two different Wifi Cards

csteamengine, 2018-02-16:
Hello, 

First off, I wanted to mention that I have tested the [Enker Wifi USB Stick](https://www.amazon.com/gp/product/B06XC6T7KB/ref=oh_aui_detailpage_o05_s02?ie=UTF8&psc=1) with the RT5370 chipset, and it works right off the bat, just plug and play. The picture is definitely a bit glitchy without wiring a dedicated 5v to the card, and the range is not great, but for $10 and little setup it was a great option, at least for testing.

That brings me to my actual question. On my [controller](https://github.com/csteamengine/heimdall-mk1/wiki/Remote-Control) I have very limited space, so I was wondering if I used this Enker wifi card on the controller, and then on the quadcopter I placed one of the bigger, long range cards, would they work together? Basically, I would keep the controller the way it is, and then put the [Alfa AWUSO36NH](https://www.amazon.com/dp/B0035APGP6/_encoding=UTF8?coliid=I10Q0XX6OCO2J1&colid=1GM19JP7K9I8P&psc=0), for example, on the drone. If something like this has never been tested before, then I would gladly test it out, and post my findings. 

Sorry if this is a duplicate, but thanks for your help!

rodizio1, 2018-02-16:
You can mix different cards. When using the wifibroadcast RC functionality you need all Atheros cards though.

The 036NH is not recommended, only about 70mw output power. I assume you want to use 5GHz? If yes, then I'd recommend the 052NH (with good cooling and wiring) as TX and one or two more 5370 for RX, it works much better with diversity.

csteamengine, 2018-02-16:
Awesome, thanks for your response! 

Since I have such limited space for RX modules, I will likely not be able to double up. I will definitely go for the 052NH then, if that is what you recommend. Thanks again for your help!

closed 2018-02-16 by csteamengine

rodizio1, 2018-02-17:
If you take them out of the case and de-solder USB and antenna connectors they are quite small. Nice controller BTW.

csteamengine, 2018-02-19:
Thank you! Wouldn't be very cool without EZ-WifiBroadcast though!