rodizio1/EZ-WifiBroadcast #96: Splash Screen Overwrite

csteamengine, 2018-02-20:
Hello again, 
I have been thinking of making my own custom splash screen, and I know there is a line in the config file to disable the splash, but I was wondering where the actual splash file is located so I could just override it. If anyone knows where that file is located, that would be awesome, thanks!

csteamengine, 2018-03-01:
Just checking back to see if anyone knows how to overwrite the splash screen that just says "EZ-WifiBroadcast 1.6 (RC3)" or something like that. Thanks in advance!

seeul8er, 2018-03-01:
Take a look at `/etc/init.d/` on your wbc-image. Somewhere in there must be a start-script that calls `wifibroadcast_status/wbc_status [some parameters]`

wbc_status is the programm that displays the text.

I do not have the image at hand but I hope this is pushes you in the right direction

tictag, 2018-03-01:
I don't think the "EZ-WifiBroadcast 1.6 (RC3)" is actually a splash screen (image), more that the code first loads a graphics code snippet then displays it as large font text. I can't remember where it is but if you clone the repository then search for the splash screen text, you'll find where it is called. For a simple update, just change the text and recompile. For a more advanced update, you'll need to know how to manipulate the graphics code (I'm guessing it's open source so docs will be available somewhere).

rodizio1, 2018-03-01:
Yes, it's the same tool that also display the status messages like "2nd display detected" etc.
It's called in /etc/init.d/raspi-config (which gets started early during boot-up, that's why it's in that script).

csteamengine, 2018-03-02:
Ok, I was able to finally locate that line of text, thank you guys for your help!

closed 2018-03-02 by csteamengine