rodizio1/EZ-WifiBroadcast #97: Modulate power vs target rssi

roque-canales, 2018-02-25:
Hello it could be good feature for EU user to real time modulate power targeting rssi level wanted.

rodizio1, 2018-03-01:
I would like to have such a feature also. Unfortunately, it's not easy. Implementing the logic for correctly in a way that makes sure that the transmit power can't go too low accidently (but without being too high all the time ...) would already require a lot of time and testing. 

And then the kernel, drivers and maybe wifi card firmware etc. need to be hacked, I have not enough deep knowledge to make that work, sorry. 

If there is somebody who has the skills to hack the wifi drivers, kernel as well as raspivid to allow for:
- dynamic transmit power
- dynamic datarate (both raspivid and the wifi card)
- some means to detect and/or control packet injection time on the card through userspace

I could improve EZ-Wifibroadcast tremendously.

rodizio1, 2018-03-31:
I'll close this now as I don't see this happening anytime soon.

closed 2018-03-31 by rodizio1