rodizio1/EZ-WifiBroadcast #98: [Not an issue] Add support for Banana Pi M2 Zero

careyer, 2018-03-05:
Hi Rodizio,

When I drank my first coffee this morning and browsed through the latest Pollin catalogue, my eyes suddenly got stuck looking at this product:
https://www.pollin.de/p/banana-pi-m2-zero-810854?utm_source=billiger.de&utm_medium=pvgl&campaign=pvgl/billiger.de&channelid=psm

Form factor of a Pi0/Pi0W but with a 
- Quad Core ARM Cortex A7 CPU H2+ 
- MALI-400 MP2

IMHO this would make for the perfect AirPi module. Same small Form Factor as the Pi0 but way more processing power.
Maybe you can support this module in one of the upcoming Releases? 
### That would be AWESOME!
Cheers! And keep up the excellent work!!!

geofrancis, 2018-03-05:
The video encoding is tied to the broadcom GPU so unless it's got an identical GPU it wont work.

seeul8er, 2018-03-05:
I guess this could work very well. However the problem is (as always) the camera module. Banana Pi only supports the OV5640 Modules/Sensors?! (Pi cam has OV5647 and IMX219). Someone should go and make a quality and latency comparison.

So many people use cameras for their projects and yet only the Raspberry Pi boards come with some useable hardware. I wonder why there is no one building a SBC (and camera modules) that work with some high quality cameras like the ones you see in modern smartphones.
Today every one talks about neural networks and image recognition but I feel like the capable hardware is either missing or way to expensive.

careyer, 2018-03-28:
@seeul8er / @rodizio1:
Got it! Wrong GPU to run raspivid. Too bad. I am wondering if it can be used as a GroundPi though? It would make a great SBC for integrating it into a small controller or receiver that fits into your shirt-bag.
(raspivid is not used on the groundside, right?). Just wondering if the kernel currently used will be compatible?

rodizio1, 2018-03-30:
This question comes up about once every few months. The short answer is, you guys are completely under-estimating the complexity of this and getting it to work on another platform. No, having a quick look at the banana pi or orange pi website and forum is not enough ;)

Here I wrote something about it: https://github.com/bortek/EZ-WifiBroadcast/issues/23#issuecomment-315685912

Or search through the rcgroups and fpv-community forum threads. Try it out if you like. I'm not going to spend any time on this.

closed 2018-04-01 by rodizio1