rodizio1/EZ-WifiBroadcast #100: Serial Telemetry

francoisrupert, 2018-03-17:
Hi Guys, nice work on keeping the ez-wbc project going. ... Simplicity of the package is great. I seem to be having some trouble with serial telemetry. the downlink is inconsistent and the uplink does not seem to be coming through complete. It seems some serial data is going out from the air-side radio from being passed on from the ground pi 3B. the radio seems fine since when i put it directly in the PC it works perfect. It seems that the parameter requests are not passed on. My config is as follows

Uplink: PC/Pad =wifi=> GroundPI ==> RFD900 ==> RF900 ==> pixracer          Does not work
Downlink Telem:  PC/Pad <= wifi= GroundPI <== RFD900 <== RF900  <== pixracer     WORKS
Downlink Video:  PC/Pad <= wifi= GroundPI <== 5.8GHz EZ-WBC  <== pi0w    WORKS

GCS = QGroundControl

Any recommendations appreciated.  I did a similar system implementing WBC (benfinitiv) and MavProxy and all worked perfect.. How is the telem uplink implemented?          

Cheers!!!

rodizio1, 2018-03-26:
Mavlink telemetry is in a not-perfect state at the moment unfortunately because of compatibilty issues with mavlink libraries (didn't know that mavlink protocol is broken-by-design, see the discussion a few pages back on the RCGroups thread if you're interested).

I'll need to write an own mavlink parser to properly fix that, which may take some time, sorry.

What should work however is using external modules, I'll look into that and see if it can be fixed quickly.

frstussy, 2018-03-26:
I'll check out the mavlink thread I was under the impression the libraries were pretty solid since so many have implemented them on various platforms. Which version are you implementing 1 or 2? and what are you using to push the UDP from wifi from the ground side to the serial on the ground side? In which file in the source do you handle the serial TX of the mavlink from ground....?? As you've probably gathered I'm opting for mavlink via RFD900 for robustness in the 900MHz band and the 5GHz for the video etc... 

I really like your implementation and the simplicity of the images and getting them running. Keep up the good work.

rodizio1, 2018-04-01:
Regarding the mavlink issues: It's rather complicated, the rx_telemetry_buf receives it, then it gets distributed to different fifos via tee and ftee, then through socat virtual serialport to mavlink-routerd and then to Missionplanner (and vice-versa). That's also why I want to write my own logic, that would mean I could get rid of socat and mavlink-routerd.

The mavlink library I'm using is the latest v1 library from mavlink.org.

Regarding the uplink via your 900Mhz modules, this was indeed an issue in the .profile script.

Can you try replacing the whole uplinktx_function in /root/.profile with this one and report back if it works?

```
function uplinktx_function {
    # wait until video is running to make sure NICS are configured
    echo
    echo -n "Waiting until video is running ..."
    VIDEORXRUNNING=0
    while [ $VIDEORXRUNNING -ne 1 ]; do
	VIDEORXRUNNING=`pidof $DISPLAY_PROGRAM | wc -w`
	sleep 1
	echo -n "."
    done
    sleep 1
    echo
    echo

    while true; do
        echo "Starting uplink telemetry transmission"
	if [ "$TELEMETRY_TRANSMISSION" == "wbc" ]; then
	    echo "telemetry transmission = wbc, starting tx_telemetry ..."
	    NICS=`ls /sys/class/net/ | nice grep -v eth0 | nice grep -v lo | nice grep -v usb | nice grep -v intwifi | nice grep -v relay | nice grep -v wifihotspot`
	    echo -n "NICS:"
	    echo $NICS
	    if [ "$TELEMETRY_UPLINK" == "mavlink" ]; then
		VSERIALPORT=/dev/pts/0
		UPLINK_TX_CMD="nice /root/wifibroadcast/tx_telemetry -p 3 -c 0 -r 2 -x 0 -d 12 -y 0"
	    else # MSP
		VSERIALPORT=/dev/pts/2
		UPLINK_TX_CMD="nice /root/wifibroadcast/tx_telemetry -p 3 -c 0 -r 2 -x 1 -d 12 -y 0"
	    fi

	    if [ "$DEBUG" == "Y" ]; then
    		nice cat $VSERIALPORT | $UPLINK_TX_CMD -z 1 $NICS 2>/wbc_tmp/telemetryupdebug.txt
	    else
    		nice cat $VSERIALPORT | $UPLINK_TX_CMD $NICS
	    fi
	else
	    echo "telemetry transmission = external, sending data to $EXTERNAL_TELEMETRY_SERIALPORT_GROUND ..."
	    nice stty -F $EXTERNAL_TELEMETRY_SERIALPORT_GROUND $EXTERNAL_TELEMETRY_SERIALPORT_GROUND_STTY_OPTIONS $EXTERNAL_TELEMETRY_SERIALPORT_GROUND_BAUDRATE
	    if [ "$TELEMETRY_UPLINK" == "mavlink" ]; then
		VSERIALPORT=/dev/pts/0
	    else # MSP
		VSERIALPORT=/dev/pts/2
	    fi
	    UPLINK_TX_CMD="$EXTERNAL_TELEMETRY_SERIALPORT_GROUND"
	    if [ "$DEBUG" == "Y" ]; then
    		nice cat $VSERIALPORT > $UPLINK_TX_CMD
	    else
    		nice cat $VSERIALPORT > $UPLINK_TX_CMD
	    fi
	fi
    	ps -ef | nice grep "cat $VSERIALPORT" | nice grep -v grep | awk '{print $2}' | xargs kill -9
    	ps -ef | nice grep "tx_telemetry -p 3" | nice grep -v grep | awk '{print $2}' | xargs kill -9
    done
}

```

The function runs on TTY10, if you connect a keyboard to the RX (without TX running so that the video is not in front of the console text) and press "alt-F10", you should be able to see if it has started correctly and is sending the data to the correct serialport.

rodizio1, 2018-04-02:
1.6RC6 with the above fix has been released, would be nice if you could report back if the up/downlink via your 900Mhz modules works now.

frstussy, 2018-04-02:
Tested with your mod of .profile yesterday and it seems to work. I will continue testing and also check the RC6 image soon. Thx

closed 2018-05-02 by rodizio1