rodizio1/EZ-WifiBroadcast #101: Help Needed: Simplifying EZ-WiFiBroadcast

tictag, 2018-03-18:
My knowledge thus far of EZ-WifiBroadcast has led me to two conclusions:

1.  Whilst the image/code contains many great features, they are all called from one central place: the .profile script and
2.  The core video transmission code, rx_rawsock and tx_rawsock, have been improved since Benitiv's original.

So I need to simplify things, right now I don't need OSD, telemetry or integration with MAVLINK, screenshots, hotspots or even interactive logins - I just need to transmit two video streams. So I've realised two things:  I'd be better off using EZ-WiFiBroadcast code and I would need to simplify .profile to meet my relatively frugal needs.

The trouble is, I really am a novice coder, no formal education whatsoever and what I know I've learned through hacking around with Arduino's and a bit of online Python; I am not clever enough to interpret .profile.

Could I ask for some help? Would you be able decompartmentalise .profile into 'core' and 'optional' code sections? I don't mind having a first stab at this if you're willing to review my efforts.

Of course, I won't need to change anything else with the image, the additional features will always be there for me when the time is right, but right now I'm being overwhelmed and feel my project will stall until I can simplify things.

vonfrank, 2018-03-22:
I am interested in this as well.. 

I have a project going, where i work on an iOS solution, and it would be cool if all features could be disabled, or some way we could pull the features required, and build our own images..

The iOS app i am working on, is simply a screen like the android version, but i want all telemetry, video etc. to be rendered at the iOS/android device, why i need to disable all other features.

rodizio1, 2018-03-26:
You guys need to get some linux and bash scripting knowledge if you want to make large changes like this. But I don't really see the point in it, you can achieve what you want with rather small changes, Tictag. I think Careyer is 99.9% there.

Vonfrank: Why change any scripts, just disable all the OSD elements if you don't want to see them on the HDMI monitor. Or don't connect a monitor at all if you're using an ios/android device ;)

iOS solution sounds great, will that be openly available in the future? You may want to have a look in the wifibroadcast_rx_status_forward_t struct in rssi_forward.c if you want to have RSSI and other data.

vonfrank, 2018-03-27:
I agree, i might be complicating it..

Yes, the ios version will be open, like the rest of the project, of course :) Have been following this project since 1.3 and really enjoy it..

rodizio1, 2018-04-02:
Thanks. Please let me know when your app is ready (or when you have a link), then I can add it to the wiki if you like.

closed 2018-04-02 by rodizio1