rodizio1/EZ-WifiBroadcast #102: USBtv007 driver

vonfrank, 2018-03-22:
Would it be possible to add a driver for USBtv007? I have a gimbal which only provides analog video out, and it would be cool if i could plug an analog to usb converter on, and have my gimbal video as a stream/secondary stream.

rodizio1, 2018-03-26:
If it's just enabling a driver in the kernel I can do that. Just did a quick google search but couldn't find which driver it is exactly or how the kernel module or config option is named, do you happen to know that?

geofrancis, 2018-03-27:
it would be better if you enabled the UVC drivers since they are universaly compatible. as far as i know the drivers should already be in raspberry pi os.

vonfrank, 2018-03-27:
@geofrancis would that work with the various easycap dongles? They come with serveral chips, the one i have is UTV007 (Fushicai)..

vonfrank, 2018-03-27:
@rodizio1, have been digging into this, and i found this guide. http://raspberry-at-home.com/video-grabber-for-raspberry-pi/

According to this, unfortunately the driver isn't a part of the default kernel module, but the driver can be found in this package: https://github.com/gkaindl/ambi-tv/archive/master.zip

If it wouldn't be an included feature, could you guide me in a direction, so i can learn, how to do it myself? I would like to learn a bit more about linux, and i would like to be able to add this feature myself aswell..

Thanks

geofrancis, 2018-03-27:
The problem with UTV007 is its only in a specific version of the easycap dongle so its a bit of a pain in the arse to find. where as with UVC you get compatibility with a lot of devices from cameras to capture cards.

rodizio1, 2018-03-30:
I'll have to agree with geofrancis, too much hassle and effort to get that special external driver working for just one (discontinued by now?) chip when there is a built-in generic uvc video driver.

The uvcvideo driver is already compiled-in in the current EZ-Wifibroadcast 1.6RC5 release, just checked it:
```
root@wifibroadcast(rw):~# modinfo uvcvideo
filename:       /lib/modules/4.9.35-v7/kernel/drivers/media/usb/uvc/uvcvideo.ko
version:        1.1.1
license:        GPL
description:    USB Video Class driver
author:         Laurent Pinchart <laurent.pinchart@ideasonboard.com>
srcversion:     FD649E242077291DE814978
```

If you want to have a stab at your usbtv007, you need to download the kernel sources and apply patches, compile it (see the /kernel/ directory in this repo for instructions and patches), install it and then compile and install the external drivers.

geofrancis, 2018-03-31:
so how would i go about trying a uvc device with wifibroadcast?

rodizio1, 2018-04-01:
I have never used such a device. I guess connecting it, checking with dmesg if it's been detected and then running v4linux-something to configure it? Sorry can't help much there.

closed 2018-04-01 by rodizio1