rodizio1/EZ-WifiBroadcast #103: 1.6 supports 34 wifi cjhips ony

gba-gbob, 2018-03-24:
As per below code if card does not fall under names  stated   it does not get put into monitor mode  and fails to transmit receive packets. Would expect either error or sensible defaults for other devices.

https://github.com/bortek/EZ-WifiBroadcast/blob/master/wifibroadcast-scripts/profile#L230
https://github.com/bortek/EZ-WifiBroadcast/blob/master/wifibroadcast-scripts/profile#L307

gba-gbob, 2018-03-26:
Hi Rodizio1, I have MediaTek MT7601 which does not get initiated but is in the wiki. I also can confirm that it works. Thanks for quick response

gba-gbob, 2018-03-26:
In addition it would be great to have ability to provide card config  files based on driver name / usb id ? in this case new card can be added by users on the fly just by creating new file, and source it from the main script.

rodizio1, 2018-04-02:
1.6RC6 has just been released, would be nice if you could report back if the mt7601 is working now.

gba-gbob, 2018-04-08:
Can confirm works fine, I cannot see changes being published in master

rodizio1, 2018-04-29:
Thanks for confirming. Sourcecode on Github will follow (sorry, I'm always behind with that). Sourcecode in the image is there.

closed 2018-04-29 by rodizio1