rodizio1/EZ-WifiBroadcast #104: The new version of Raspberry Pi 3b+

Matelas, 2018-03-25:
EZ-WifiBroadcast does not start on Raspberry Pi 3 model B+ WiFi Dual Band
https://www.raspberrypi.org/products/raspberry-pi-3-model-b-plus/
Is there any update planned in the nearest future?

rodizio1, 2018-03-26:
It has just been released, there are different ethernet, usb and wifi chips on it, and from looking through the posts on the Raspberry Github and Raspberry Forum, there still are a lot of issues. Will probably take another month or two until the Raspberry foundation has ironed-out these issues and then I need to find time to upgrade the kernel and firmware, so I wouldn't wait on it if you want to fly before summer.

matiaspl, 2018-04-14:
Seems that there's now a working Pi3 B+ packet injecting wifi driver: https://www.rtl-sdr.com/nexmon-sdr-using-the-wifi-chip-on-a-raspberry-pi-3b-as-a-tx-capable-sdr/

seeul8er, 2018-04-14:
Saw that too. Looks very promising if they can get reception working as well. But I do not know if the range/power of the onboard chip is worth the trouble.

radlxau, 2018-07-29:
Has there been any progress on this at all got a bunch of PI 3B+ and would like to see if this can work as trying to find a USB dongle that will do it is becoming challenging

rodizio1, 2018-11-07:
>Seems that there's now a working Pi3 B+ packet injecting wifi driver: https://www.rtl-sdr.com/nexmon-sdr-using-the-wifi-chip-on-a-raspberry-pi-3b-as-a-tx-capable-sdr/

This has nothing to do with packet injectiion, it's about using the Pi Wifi Chip as a software defined radio, which is something completely different and not at all suitable for our kind of application.

If we want to use the Raspberry onboard Chip for transmitting, we first need to fix this:
https://github.com/seemoo-lab/nexmon/issues/101

Another general issue with the Pi3B+ is, that it seems to crash under certain conditions and it's still unclear what the exact cause is, if there is more than one cause (CPU/RAM overclocking, Ethernet or USB Chip or firmware issues ...) and if it has been fixed. The Pi foundation seems to play this issue down.

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=208821&hilit=memtester#p1290869

rodizio1, 2019-01-11:
Current state is: The current Kernel 4.14 gives various issues with USB and Wifi cards, so we cannot yet use that kernel. An alternative would be the 4.9.80 kernel from Rasbian 2018-03-13, however, this was the first release which included support for the Pi3B+ and had various issues (unstabilities due to RAM,CPU overclocking, Ethernet and Wifi issues) according to the Raspberry forum, so this one also needs thorough testing.

MatthewFraley, 2019-01-27:
I apologize if the answer is given elsewhere but would using a usb wifi dongle on the pi3B+ allow it to be used with EZ-Wifibroadcast? If so how would I need to go about this?