rodizio1/EZ-WifiBroadcast #105: add compatibility to AWUS036ACH

roque-canales, 2018-03-26:
awus052nh no exist anymore, 

AWUS036ACH is now injection compatible, there's is any body that can help us for make that this 5ghz board work correctly with ezw?

rodizio1, 2018-03-26:
I'll look into that some time (but it can take a few months, sorry).

rodizio1, 2018-03-30:
Did some googling.

The AWUS036ACH really looks promising, finally a card with high power and two transmit/receive chains. It has the same amp on it as the 036NHA, so should in theory give the same output power. The RTL8812AU chipset also supports STBC and LDPC which should increase range and robustness of the link in theory.

I did some search about monitor mode/injection and drivers, there are like 10 or more different forks of various Realtek driver version available on Github, the most promising driver seems to be the rtl8812au from the aicrack-ng guys (https://github.com/aircrack-ng/rtl8812au).

However, to support the new Raspberry Pi 3B+ in the future, we'll have to wait until the Raspberry Kernel and firmware has stabilized. From what I read on the Raspberry forum, they also plan to transition to kernel 4.14 in the near future so I think it doesn't really make much sense putting work in getting this driver to work with kernel 4.9.

If somebody wants to have a try at this themselseves in the meantime, quick "howto":
- Get the latest Raspberry firmware from the Raspberry Github repo, copy over to /boot
- Get the latest 4.14 Kernel sources from the Raspberry Github repo
- Apply the EZ-Wifibroadcast kernel patches, and possibly adapt them to 4.14 if they don't work
- Apply the EZ-Wifibroadcast kernel config with "make oldconfig" and possibly adapt it to 4.14
- Build and install the kernel
- Download the rt8812au drivers from aircrack-ng github repo
- Compile and install the drivers
- Do a lot of testing to see if it really works and doesn't drop packets or is unstable/unreliable or does whatever other weird things
- Test different channels to see if all are working
- Check transmit power, make sure it's not too high and not too low, find a way to set custom transmit power

roque-canales, 2018-03-30:
Hello Rodizio,

Thank you so much for this feedback.

Could be possible that you compile actual version of EZ WBC with theses modifications for thoses who want to test rtl8812AU boards ?

For my part I can do lot tests (power output, range on some channels, range with different baudrate setings) when you release out quite firmware and share it whith the community.

rodizio1, 2018-04-01:
I don't have any realtek 8812 cards to test with, so doesn't make much sense. I'll give it a try when I got some.

roque-canales, 2018-04-01:
Rodizio,

give me your adress and I send you two boards

my email:
kuspower @ live . fr

nrodichenko, 2018-04-27:
Dear Rodizio,

Regarding your comment about rpi3b+ - would it be possible to skip right to driver compilation & testing on rpi3b+ez-wbc1.6? Is rpi3b+ somehow necessary?

roque-canales, 2018-04-27:
How to apply the EZ-Wifibroadcast kernel patches ?

nsimon1, 2018-05-01:
@rodizio1 
Raspbian has now been moved up to kernel 4.14, I have ordered two of the 036NHA dongles and should hopefully get them by the end of next week, then I will take a shot at getting it up and running using your recommendations and can also help testing then. 

If any one else has already made progress with Kernel 4.14 I would appreciate the head start. 

Thanks!

wstoettinger, 2018-05-06:
Hey @rodizio1,
I managed to compile and install the driver for AWUS036ACH as well as the wifibraodcast software on RPI 3 B+. The problem now is, that I don't get it to listen to my RPI 3 B (no plus) with my AWUS036NHA. 

I followed the instructions here to make the prepare the kernel for the build: 
https://raspberrypi.stackexchange.com/questions/50240/missing-build-file-when-building-for-rtl8812au-driver

Then I cloned the driver fork from aircrack-ng https://github.com/aircrack-ng/rtl8812au and i changed the following lines in the makefile of the driver:

```
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = y
```
dkms install didn't work so I chose the normal way via make and make install.

Then I continued with some recommendations from another issue here: https://github.com/bortek/EZ-WifiBroadcast/issues/108

i set up the monitor mode and listened to channel 13
```
sudo service network-manager stop && \
sudo ifconfig wlan1 down && \
sudo iw dev wlan1 set monitor otherbss fcsfail && \
sudo ifconfig wlan1 up && \
sudo iwconfig wlan1 channel 13
```
so far so good. When I start to to listen now, I don't get anything
```
sudo ./sharedmem_init_rx
sudo ./rx -b 8 -r 4 -f 1024 wlan1
```
So my guess was, that either the Tx is not working or not sending on channel 13 (although I do see that I get packets on the Rx with ifconfig).

I hope this helps somebody and I'd appreciate some help on how to check if the Tx is working correctly ...  when I plug in a screen it always starts as Rx. 

thanks in advance!

roque-canales, 2018-05-17:
Hello  @wstoettinger 
Could you please share on wetransfer your ezwbc image for we test the AWUS036ACH ?

like the we can share our result with all?

Thank you so much!

geofrancis, 2018-05-29:
I think this needs to be a priority, compatible wifi cards are drying up quickly, with only the ALFA AWUS036NHA and AWUS051NH  still readily available.

ubiquity wifistation cards are not made any more 
AWUS052NH are not made any more
TPLink TL-WN722N V1 are not made any more

wstoettinger, 2018-05-29:
@roque-canales 
I'm sorry, I currently don't have the time to work on that. 
However I think that my setup is easy to reproduce with the detailed instructions I gave. 

Let me know, if you run into problems!

Yes21, 2018-06-02:
Hello,
I want to test the AWUS036ACH for rx (Pi 2 B) and the AWUS036AC for tx (Pi zero 1.3).
I've first checked that the 2 cards were working on my laptop wiyh Ubuntu 18.04, and they do very well after having installed the rtl8812au-dkms driver from ubuntu.
Then I've tried them with the Pis and ezwbc 1.6 rc6, without modifying anything. On the rx side (Pi 2) I think the cards are recognized and initialized, because the blue led are blinking after a few seconds. For now I can't log on the Pi with ssh, because I'm waiting for a swith.
But on the tx side both cards didn't work : no blue led. I think it's a problem with the micro usb port of the Pi zero, and with its output power. I've tried to power the AWUS036AC directly, but no success.
For the next step I will try to solder the usb D+ and D- directly on the Pi zero.
I've also tested a small wifi "Edup" dongle on the TX, and could receive a very bad image on the rx. I thnk it's normal because the dongle has no antenna and very low power.

I will give you later more news.

ps : it's my first experiences with ezwbc !!
So this confirm that all seems to work for the rx, and the problem is the connection of the AWUS036AC with the Pi zero.

rodizio1, 2018-06-29:
Some more infos here (in russion, can be translated with google translate though):

http://forum.rcdesign.ru/f90/thread515083-7.html#post7421810
http://forum.rcdesign.ru/f90/thread515083-7.html#post7426801
http://forum.rcdesign.ru/f90/thread515083-9.html#post7446277

Yes21, 2018-07-07:
To go further in my tests, I've try to buy a compatible 5 Ghz card for the TX. The last available one is (was) the TP-Link TL-WDN3200 (RT5572 chipset). I've ordered one, but I received a TP-Link Archer T4U (8812AU) instead of it, because the TL-WDN3200 isn't anymore available !
So today, no more 5 Ghz wifi cards usable with EZ-WBC are any more available.

Today almost all dual band cards have a 8812AU (or 8814AU) chipset.
I've tried the AWUS036AC and AWUS036ACH, and I can confirm that they are working "out of the box" as RX with 1.6RC6, because you had added 8812AU support in 1.6RC1 ("_Support for RTL8812AU cards added (only RX, not tested, for experimenting)_"). That's already a good thing !

I have now an AWUS1900, and will soon try to install the aircrack-ng 8814AU driver on my RX Pi.

But on the TX side, I don't know what to do to have 8812AU support ...

My "dream" is to have a working 5 Ghz EZ-WBC system working with, as TX a Pi0 with a AWUS036ACH (or AWUS036AC because it's smaller), and as RX a Pi2B with an AWUS1900.

I hope I will see it in the near future, and do what I can to help ...

roque-canales, 2018-07-21:
Does anybody found solution?

htcohio, 2018-08-10:
36ACH should be arriving tomorrow.  I don't know anything about coding and the primary reason I got this was to have 5G hotspot for the various Android apps. 

I have tried a couple non wfb 5G dongles for the hotspot on the RX side and they are not recognized.

My Hope Is but that I can use the 5G Mac address for the hotspot to receive video wirelessly and then also have 3 rx atheros cards running 2.3/2.4.

I kind of assumed that everything was working with the 5G side but after reading this it doesn't sound like that's the case?

This was a $50 Wi-Fi card and my thinking was that it could be used in the future on the TX side in the air for 2. 4/5. 8 diversity.

Any comments or feedback would be appreciated.

roque-canales, 2018-08-24:
@htcohio, hello, did you reach to upgrade EZWBC for use 8812AU ?

Yes21, 2018-09-08:
I can now confirm that the image of this post ([http://forum.rcdesign.ru/f90/thread515083-9.html#post7446277](http://forum.rcdesign.ru/f90/thread515083-9.html#post7446277)) is working very well with 8812au and 8814au.
Awus036ac, awus036ach and awus1900 are all working as TX and RX, on Pi zero, Pi2b/3b (I've no Pi3b+)

bortek, 2018-11-02:
I am trying to clean up the clutter @Yes21 , do you know if this is still relevant or can we close it ?

Yes21, 2018-11-02:
I will soon close it when I will be sure that rtl8812/14 are really supported ...

Yes21, 2018-11-04:
With the v5.2.20 branch of the driver, I always have  ZERO packetloss.

careyer, 2018-11-04:
> And with this friver, I can make work an AWUS1900 on the groundPi and a AWUS036ACH on the airPi ...

Want to give this image another try. Which stages do I have to repeat from the last one I build several days ago? All stages or just certain ones?

RespawnDespair, 2018-11-04:
Due to the many modifications in the kernel stage it would probably be better to redo everything.
Actually, a clean git clone in a new folder would probably be advisable.
Once the kernel issues have been fixed you'll probably not touch step 2 for a long while.

careyer, 2018-11-04:
Thank you very much... I will exactly do so. My house is rather getting cold - this is a welcome excuse for letting the CPU crunch some numbers this afternoon. 😋 - will report back

RespawnDespair, 2018-11-04:
Please hold one moment, i'm changing some thing around

RespawnDespair, 2018-11-04:
Okay, if you get the latest it at least starts the kernel build again.
No guarantees beyond that 😄

roque-canales, 2018-11-12:
hello after check AWUS036ACH is only 21dbm, twpower_ralink no change anything…

How to set up the tx power on this board ?

Yes21, 2018-11-12:
`sudo iw wlan0 set txpower fixed 4000`

-> 40 db

roque-canales, 2018-11-12:
thank you, is your interface the mac adress?

Yes21, 2018-11-12:
Yes, without ":" 
Example "00c0ca9679c8"