rodizio1/EZ-WifiBroadcast #108: Use a laptop as rx

maxmvanderwel, 2018-04-04:
Is it possible to use a Linux laptop as rx and how?

maxmvanderwel, 2018-04-07:
> I have never tried it myself so cannot really help much. From the top of
> my head, I'd say you probably have to remove the wiringPi library from the
> Makefile (and tracker.c which uses it) to make it run on a PC. Also you
> need to run the sharedmem_init_rx program before to initialize the shared
> memory.

I’m gonna try that.
Thanks for your help

user1321, 2018-04-09:
You can use PC for both RX and TX role. But  take attention at video player at RX side. For example vlc gives me around 1- 2 second of latency , and nothing less. With same settings at TX side mplayer shows 0.25 sec or 250 ms of latency .

user1321, 2018-04-12:
Can confirm that mavlink two way control work too.
PC1 (Ground Control) Mission Planner (udp 14550) <-> Ethernet <-> PC2 (Ground ReciveTransfer station) mavlink-routerd (rx_rc_telemetry_buf -p 1 . + tx_telemetry -p 3 ...) < Wireless link> PC3 (Air station) -> APM.
It took some time to analyze and understand how./profile scripts works even if all work already done :) by Rodizio1 and community (probably) ...(Thx for great job btw)
If it will be useful for someone I can public short version of commands needed to run that setup (tiny version of ./profile without RP dependency to just run mavlink connection ).

moharam-dev, 2018-04-13:
@rodizio1 Thank you for this really cool FPV.

I'm trying to do the same thing as maxmvanderwel since I want to run OpenCV on RX. My tx is a raspberry pi zero + RT5370 and working perfect. My RX is ubuntu 16.04 using the same wifi adapter as TX (i.e. RT5370).

Cloned, modified the make file, and removed the tracker and everything is compiling successfully.
I didn't get what you mean by sharedmem_init_rx program? do you mean the rx_status executable? is there a specific order of execution?

Thanks

moharam-dev, 2018-04-14:
Changed my setup a little bit for easier troubleshooting. Following my progress so far:
- 2 Ubuntu pc's. TX, and RX, Both have cheap RT5370 USB wifi adapter.
- Both adapters are running in monitor mode.
- TX: Sending dummy data (/dev/zero). Sending data using TX is a success (verified that using WireShark on RX)
- RX:  no data received by wifibroadcast so far! Still don't know what is going on. I debugged the code and notices that "select()" always returning zero (timeout).

Any help will be really appreciated :)

moharam-dev, 2018-04-14:
Thanks seeul8er, I successfully managed to send and receive using dummy data. It was as simple as making sure the TX and RX args are matched :)

Following my changes and steps so far:

1) Makefile changes:
- Removed "-lwiringPi" from linker options.
- Removed all "Tracker" compilation reference.
rx.c:
- Modified for debugging purposes (added a statement to print to stdout when a packet received)

2) Build executables
Make tx rx

3) Run on two separate machines as root:
TX command: 
cat /dev/zero | ./tx -d 6 -b 8 -r 4 -f 1024 -t 1 -y 0 <wlan_tx_nic>
RX command:
./rx -p 0 -t 1 -d 6 -b 8 -r 4 -f 1024  <wlan_rx_nic>

However when I switched the TX with Raspberry Pi Zero which has wifibroadcast image installed I didn't receive my debugging messages on RX. 

How would I know the args values passed to tx_rawsocket on RPZ?

Will try to decode this using wireshark. but this will need some time of course :(

user1321, 2018-04-14:
Take a look at wifi channel. Maybe it not set to same channel at TX and RX?
Just in case my code to start video:
Maybe it will help you...
1. Init wifi card ( at both sides )
service network-manager stop
ifconfig wlo1 down
iw dev wlo1 set monitor otherbss fcsfail
ifconfig wlo1 up
iwconfig wlo1 channel 6

Air side:
1. init shared memory
./sharedmem_init_tx

2. Transfer data from web cam ( in my case /dev/video1 ) 
ffmpeg -f v4l2 -i /dev/video1 -vcodec libx264 -pix_fmt yuv420p -tune zerolatency -preset ultrafast -f mpegts pipe: | tx_rawsock -p 0 -b 8 -r 4 -f 1024 -t 1 -d 36 -y 0 wlan0

Ground side:
1. Init wifi card ( at both sides )
service network-manager stop
ifconfig wlo1 down
iw dev wlo1 set monitor otherbss fcsfail
ifconfig wlo1 up
iwconfig wlo1 channel 6

2 Init shared mem
./sharedmem_init_rx

3. Recive data from web cam and pipe it to mplayer 
./rx -b 8 -r 4 -f 1024 wlo1 | mplayer -demuxer +mpegts -framedrop -benchmark -

... next post about MavLink and rssi

moharam-dev, 2018-04-14:
Great. Thanks user1321. Will try this right away.

moharam-dev, 2018-04-14:
Now I'm able to see my debugging messages on RX. TX RPZ <-> RX UBUNTU is working. The missing part was to specify channel 13 on RX (i.e. iwconfig wlo1 channel 13... Thanks user1321). Wireshark started to capture the TX traffic successfully and RX is showing my messages :)

Tomorrow I will work on the video part.

user1321, 2018-04-14:
One more thing. A little mistake with name of wifi adapter, wlo and wlan0... but please, change it to name  of yours wlan . 
Next code responsible for control APM 2.8 using MavLink protocol via wifi 
Air side:
1,2 Init wifi card, init shared mem. (done at code above. Just one time )
(Don`t remember, maybe it needed to init both rx and tx: ./sharedmem_init_tx + ./sharedmem_init_rx for mavlink and rssi )
#MavLink Control (its two way):
3. Configure APM USB port ( not sure if it correct )
stty -F /dev/ttyUSB1

4.#Recive data from ground station:
./rx_rc_telemetry -p 3 -o 0 -b 115200 -s /dev/ttyUSB1 -r 1 wlan0
4. Read data From APM 2.8 (USB port) and send it to ground stantion:
cat /dev/ttyUSB1 | ./tx_telemetry -p 1 -c 0 -r 2 -x 0 -d 12 -y 0 -z 1 wlan0

Ground station:
1,2. Init wifi card (done at firs post) + shared mem init
3. Create virtual ports
socat -d -d pty,raw,echo=0,link=/dev/ttyS20 pty,raw,echo=0,link=/dev/ttyS21 &
4. Configure
stty -F /dev/ttyS20 -echo 115200
stty -F /dev/ttyS21 -echo 115200
5.  Run Mav Link forward. From virtual sireal port ttyS21 to static IP address (in my case 192.168.1.44 can be localhost) At 192.168.1.44 I started a Mission Planner that accept incoming UDP to port 14550 
mavlink-routerd -e 192.168.1.44:14550 /dev/ttyS21:115200 &  
6. Accept data from air and write it to on end of virtual sireal port.
TELEMETRY_RX_CMD="/home/myuser/wifibroadcast/rx_rc_telemetry_buf -p 1 -o 1 -r 99"
$TELEMETRY_RX_CMD wlo1 > /dev/ttyS20 &
7. Read data from virtual sireal port ttyS20 and send it to Air . ttyS20 recive data from command 5 (ttyS21 -> ttyS20 )
VSERIALPORT=/dev/ttyS20
UPLINK_TX_CMD="nice /home/myuser/wifibroadcast/tx_telemetry -p 3 -c 0 -r 2 -x 0 -d 12 -y 0"
cat $VSERIALPORT | $UPLINK_TX_CMD wlo1 & 
_________________________________________________________
Something like that. 
Sorry for gramma, its a quick post.

user1321, 2018-04-14:
and one more more...) If you will run ./rx_rc_telemerty (1.6 RC6 ) for mavlink, there is a chance that you will get "Segmentation Fault" error. To fix it do:
Change rx_rc_telemerty.c at line ~600 "int i;" to "int i = 0;"
https://github.com/bortek/EZ-WifiBroadcast/issues/111

moharam-dev, 2018-04-16:
OK, I spent two days trying to make this work on my PC and I failed so far :(

- I tried initially to use RPZ as TX with the wifibroadcast image. I can receive the packets on RX but it keeps crashing in first packet processing. still trying to find out about sharedmem_init_rx program which I don't know what it is exactly or where is it in the code?!!

- Switched to @user1321 suggestion. installed raspbian on RPZ, loaded the v4l2 driver to get a gain access to /dev/video0. and tx seems to work fine. on the RX side i still unable to see the video and RX keeps warning about performance issues and no video played at the end (I can see that this method is so slow when comparing the statistics reported by the Wifibroadcast image and this method).

I'm completely blocked and have no idea what to do. any ideas?

user1321, 2018-04-16:
Just in case I add archive with source`s and files compiled under Ubuntu.
Out of uname -a:
 #143-Ubuntu SMP Mon Apr 2 16:08:24 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

[Ubuntu64WifiBroadCast.tar.gz](https://github.com/bortek/EZ-WifiBroadcast/files/1915047/Ubuntu64WifiBroadCast.tar.gz)

Ps. If my suggestion about the wrong version is incorrect, please, add command that crash add error`s stdout.
Thank you.

user1321, 2018-04-16:
In any case, As I said before, video and mavlink for fine in case if transfer from Linux based PC to Linux based PC.
I think with RPi to PC not a problem, just need to find whats wrong.
All I want to say, I know good this feeling when you can a day break head in to wall to find a little mistake with one command )

user1321, 2018-04-16:
Please," correct me if I am wrong.  but, you can't get latest (for today 16.04.2018 - version 1.6 RC6 ) by git clone "branch master" . I got source files from "EZ-Wifibroadcast-1.6RC6.zip" -> unzip -> mount image -> copy files from "/root/*  .. Maybe this causes misunderstand.

kdoubner, 2018-04-18:
Hi, finally I have stream from rpi to my notebook. Thank you @user1321 very much. Please can you describe how do you compile source files under Ubuntu? I try with information above, but no luck.
Someone to start port on x64? ;-)

moharam-dev, 2018-04-18:
Thanks @user1321 everything is working with this version of code you uploaded and everything is working like a charm :) I can receive the RPZ streaming on my Ubuntu 18.04 beta laptop.

my steps using your provided code:
- prepare the wifi adapter as above (except for the channel, I used channel 13 since it is the one selected by RPZ).
- from within your provided source code:
- make clean
- make 
switch to root (sudo -s)
./sharedmem_init_rx
./rx -b 8 -r 4 -f 1024 <wifi-adapter-> | gst-launch-1.0 -v fdsrc ! h264parse ! avdec_h264 ! xvimagesink sync=false

and then the video window popped up with the video :)

may I ask why the source is different than the one on the official git repo?

careyer, 2018-04-18:
It would be awesome if someone could build a ISO Image for a Ubuntu based RX and maybe offer it to Rodizio so that he can make it available here for download as well? That would be awesome.

moharam-dev, 2018-04-19:
I'm thinking of a privileged docker that can run the RX full stack and stream the received video out of docker to a udp port that can be consumed by the external host via gstreamer or whatsoever. 

i.e the docker will do the following: 
- Install all the required packages
- copy and build the sources locally
- Acquire the wifi adapter, and configure it.
- Run RX and stream to gstreamer pipeline that broadcast to a local address udp port.

I will try this and see what will result.

moharam-dev, 2018-04-21:
well, I tried the docker but it is obvious the docker performance is not strong enough to handle the processing required.

here it is for if anyone interested:

```
FROM ubuntu:16.04
ENV DISPLAY :0

ADD wifibroadcast /wifibroadcast/src

RUN apt update && \
    apt install -y \
    net-tools \
    iw \
    libpcap-dev \
    build-essential \
    wireless-tools \
    gstreamer1.0*

RUN cd /wifibroadcast/src && make 

ADD startup.sh /bin/startup.sh
RUN chmod +x /bin/startup.sh 
ENTRYPOINT /bin/startup.sh
```

startup.sh
```#! /bin/bash

echo "working on : $WLAN" 
ifconfig $WLAN down && \
    iw dev $WLAN set monitor otherbss fcsfail && \
    ifconfig $WLAN up && \
    iwconfig $WLAN channel 13

#gst-launch-1.0 --help
/wifibroadcast/src/sharedmem_init_rx
/wifibroadcast/src/rx -d 36 -b 8 -r 4 -f 1024 $WLAN
```
Build the image:
```
docker build -t wifibroadcast:latest .
```

Run it
```
docker run -it --net host -e WLAN=<<your wlan id>> wifibroadcast:latest bash | gst-launch-1.0 -v fdsrc ! h264parse ! avdec_h264 ! xvimagesinksync=false
```

moharam-dev, 2018-04-21:
I think a small shell script will be fine to install dependencies, build, configure, and run RX locally on a linux box.

kdoubner, 2018-04-23:
Hi,
can I ask, where do you get `/wifibroadcast/src`? From git? Image? Other source? 
Does anybody have OSD working on linux? Thank you :-)

rodizio1, 2018-04-29:
Sorry the source on Github is not up to date, you'll always find the latest source in the image though

madhavajay, 2018-07-25:
@rodizio1 I downloaded the latest stable and opened the img, I dont understand where the source is. I dont see anything that looks like src in the image. Do I need the pre-release version?

Can someone just push the latest code to a branch on git, i mean... why are we collaborating with zip files on Github?

loopdude, 2018-08-21:
An IMG of a working Ubuntu RX would be awesome!

madhavajay, 2018-08-23:
I found the source in the /root folder of the Linux Partition of the SD Card.
There is also a lot of great stuff here:
https://github.com/RespawnDespair/wifibroadcast-base

The source and additionally other repos from @RespawnDespair with scripts.
I have the ./rx working on Ubuntu 18 compiled from the above source by simply running this patch:
```
diff --git a/Makefile b/Makefile
index 88b8408..5980252 100644
--- a/Makefile
+++ b/Makefile
@@ -1,7 +1,7 @@
-LDFLAGS=-lrt -lpcap -lwiringPi
+LDFLAGS=-lrt -lpcap
 CPPFLAGS=-Wall -D _GNU_SOURCE

-all: rx_rc_telemetry_buf rx rx_rc_telemetry rssirx rssitx tx_rawsock tx_telemetry tx_measure rx_status tracker rssilogger syslogger channelscan check_alive rssi_forward wifiscan wifibackgroundscan sharedmem_init_rx sharedmem_init_tx
+all: rx_rc_telemetry_buf rx rx_rc_telemetry rssirx rssitx tx_rawsock tx_telemetry tx_measure rx_status rssilogger syslogger channelscan check_alive rssi_forward wifiscan wifibackgroundscan sharedmem_init_rx sharedmem_init_tx

 %.o: %.c
 	gcc -c -o $@ $< $(CPPFLAGS)
```

Unfortunately theres some instability where it fails to receive sometimes and im still trying to hunt down the issue. If anyone else knows anything let me know.

shyam98, 2018-08-26:
Hi, I'm new to wifibroadcast (and linux/ coding in general). I also wanted to have my setup like this to run opencv from my ubuntu laptop on the RX. I'm pretty lost in this thread as to installing the program on ubuntu. I've downloaded and installed the package from user1321 and followed instructions from moharam-dev. It does not let me make in the wbc directory. Also quite confused on "preparing the wifi adapter on both sides" I followed the instructions on ubuntu, not sure how to do it on the RPi. Sorry if there is something obvious that I am missing I have tried hard to follow this thread to the t.

shyam98, 2018-09-06:
Does anyone know how to get to the configuration file for @user1321 's code? I want to use the 2.3ghz bands on my Atheros cards but don't know how to set the proper channel on the receiving end.

madhavajay, 2018-09-08:
I made this bash script, save it as a file called setwifi.sh and then chmod +x so it can run, then run it as sudo. 5.7 Ghz is 5700 so 2.3Ghz is probably 2300, so you would run setwifi.sh 2300.
This script grabs the first WiFi adapter name so if you have more than 1 you will need to change it or just run the commands manually.

```
#!/bin/bash

if [[ -z "$1" ]]; then
	echo 'Frequency required, try setwifi.sh 5700';
else
	WLAN_FREQ=$1
	WLAN=`iw dev | awk '$1=="Interface"{print $2}'`
	service network-manager stop
	ifconfig $WLAN down
	iw dev $WLAN set monitor otherbss fcsfail
	ifconfig $WLAN up
	iw $WLAN set freq $WLAN_FREQ
fi
```

user1321, 2018-09-12:
@shyam98 , 

Hi guys!.
If someone still interested in using PC as WFBC receiver or transmitter, you can check my post at rcgroup about it.
In short in allow to receive video from RPI camera+receive video from IP camara via UDP, send two-way mavlink messages to pixhawk.
Link to rcgroup:
https://www.rcgroups.com/forums/showpost.php?p=40029102&postcount=5375
link to YouTube:
https://www.youtube.com/watch?v=sevEihOZncQ&feature=youtu.be
https://www.youtube.com/watch?v=a0qyPxrWKIg

LinkTo google drive:
https://drive.google.com/file/d/1JRSdcg8JY8_02Exa0AYRVwa3EKi4t9EV/view
https://drive.google.com/file/d/1ojG0juABSrutECBcRv7At_-FmtdUkLm4/view

P.S. As for RPi image - its was customized to allow IP camera work too. But, probably only for RPi camera you may try original. Also, if you interested in this, read rcgroup post.
Thx.
Hope thats help to someone.

user1321, 2018-09-16:
Hi!

 I have limited access to PC now (vacation), i will have full access to it on Wednesday and i will share PC image.

in short, to get lower latency for ip camera ( same as RPi camera, around 200-300 ms at PC) you may use project
https://github.com/svpcom/wifibroadcast
 for transfer video. 

it allow to transfer udp packets with  rtsp headers. that's just for info, or you may wait untill Wednesday and. I will upload it.

Thx.

bverissimo, 2018-09-16:
Thank you for replying.
I Will wait to test your image.
Have a nice vacations.

Thanks
Bernardino

user1321, 2018-09-19:
How it work:
RPi Airside:
1. Initially IP camera script launched from /root/.profile script (init_profile_script.PNG)
2. File that started named: "/root/cameracontrol/IPCamera/StartTransferIfOnline" ( print screen: StartIfUp.PNG )
It do next things:
Configure ethernet interface eth0 with static IP address 192.168.1.215
And start continuously ping IP address 192.168.1.10 (IP Camera in my case)
If it ping it successfully, it wait for 10 seconds to let IP Camera boot completely and start 2 more scripts.
First one responsible for read video data from ip camera and send it to local (RPi AIR ) udp port 5600 and Second responsible for transfer data via WiFi (radio link) to PC.
Please, look at print screen in attachment.
At your side, you may change IP address of ip camera if it not set to 192.168.1.10 from factory.
And change RTSP video request url.
So, please, find out IP and RTSP url for your camera and change it inside next files if it not same as mine:
/root/cameracontrol/IPCamera/StartTransferIfOnline.sh
/root/cameracontrol/IPCamera/startVideoCapture.sh

also, take attention at .profile script, as you may see it start IP camera checker script after main RPi camera will be initialized. That mean that you have to connect RPi camera also to let start IP camera or you may change inside .profile script place from where IP camera checker script will be launched to avoid using of RPi camera. (I use both)

Url to LiveCD+RPi image+print screens:
https://drive.google.com/open?id=11-jyhXLQnWHQr3q6BbOll5Y8j8FUcgG-

bverissimo, 2018-09-20:
Thank you so much for this amazing work.
I will give it a try, and send you feedback :+1: 

Thanks
B

bhamilton1co, 2018-09-20:
Thanks user1321 !

bverissimo, 2018-09-27:
Hello user1321,

Is it possible to Run it on a sdcard ? Whats the software to create the bootable média? Iam using Rufus but got a error about the version of my software.

Thank you

closed 2018-11-02 by bortek

andrewdreww, 2019-11-23:
> Can confirm that mavlink two way control work too.
> PC1 (Ground Control) Mission Planner (udp 14550) <-> Ethernet <-> PC2 (Ground ReciveTransfer station) mavlink-routerd (rx_rc_telemetry_buf -p 1 . + [...]

may i ask you something sir?

user1321, 2019-11-23:
Hi.
You can try, but take in mind that from last post pass more than a year and many thing was forgotten.
In any case I suggest you to take a look at OpenHD.
It have way better support.

andrewdreww, 2019-11-23:
> Hi.
> You can try, but take in mind that from last post pass more than a year and many thing was forgotten.
> In any case I suggest you to take a look at OpenHD.
> It have way better support.

thanks for your suggestion sir,
can i got your email or contact sir? i want to ask about it. would you please?

user1321, 2019-11-23:
Sure,
delme643@ukr.net

VU2ATX, 2022-06-26:
> @shyam98 , @loopdude , ...
> 
> Hi guys!. If someone still interested in using PC as WFBC receiver or transmitter, you can check my post at rcgroup about it. In short in allow to receive video from RPI camera+receive video from IP camara via UDP, send two-way mavlink messages to pixhawk. Link to rcgroup: https://www.rcgroups.com/forums/showpost.php?p=40029102&postcount=5375 link to YouTube: https://www.youtube.com/watch?v=sevEihOZncQ&feature=youtu.be https://www.youtube.com/watch?v=a0qyPxrWKIg
> 
> LinkTo google drive: https://drive.google.com/file/d/1JRSdcg8JY8_02Exa0AYRVwa3EKi4t9EV/view https://drive.google.com/file/d/1ojG0juABSrutECBcRv7At_-FmtdUkLm4/view
> 
> P.S. As for RPi image - its was customized to allow IP camera work too. But, probably only for RPi camera you may try original. Also, if you interested in this, read rcgroup post. Thx. Hope thats help to someone.
> 
> Ps2. if someone interested in IP camera, not in RPi camera, say it, I will upload to gdrive new image with lower latency for IP camera. (I just don`t want to waste time if it not needed by anyone)

Hi, what is the sudo password of LiveCD_with_IP_CAMERA.iso ?