rodizio1/EZ-WifiBroadcast #112: OSD Add proximity sensor and rangefinder mesurements

roque-canales, 2018-04-15:


roque-canales, 2018-04-30:
Hello Rodizio, 

here you have description of mavlink proximity messages:
http://mavlink.org/messages/common#DISTANCE_SENSOR

geofrancis, 2018-04-30:
this would be good as long as it can also show rangefinder altitude.

rodizio1, 2018-04-30:
Regarding obstacle avoidance: Just displaying 8 numbers probably doesn't make too much sense, it'd need some kind of graphical display and additional logic to make something turn red or warn when things come too near, that would be too much work and I don't even know how to do that.

Displaying the rangefinder distance should be no problem I guess.

roque-canales, 2018-05-25:
ok, so Don't upgrade OSD, but please upgrade EZWBC for it relay DISTANCE SENSOR MSG and RANGEFINDER masg to mission planner.

closed 2018-11-02 by bortek