rodizio1/EZ-WifiBroadcast #115: Not starting after apt-get upgrade

andrew-grischenko, 2018-04-27:
I've been having issues with installing NodeJS on the RX device, solved it by doing **sudo apt-get upgrade**, but that broke the EZWifiBroadcast. Now after boot it gives error:
```
Starting RX ...
Unable to open xxxxxxx: That device is not up
ERROR: couldn't start RX ...
```

Tried on RC5 and RC6

What can I do? I need either make Node.js working on original image or make the upgrade work. 
Appreciate any help with this issue.

andrew-grischenko, 2018-04-27:
So, what I found so far is the this command gives me 2 NICS:
```
 ls /sys/class/net/ | nice grep -v eth0 | nice grep -v lo | nice grep -v usb
a021b7cc0b32
b827ebc1e782
```
The error is about b827ebc1e782, so I check both:
```
iwconfig a021b7cc0b32 | grep a021b7cc0b32
a021b7cc0b32  IEEE 802.11  Mode:Monitor  Frequency:2.472 GHz  Tx-Power=20 dBm  

iwconfig b827ebc1e782 | grep b827ebc1e782
b827ebc1e782  IEEE 802.11  ESSID:off/any 

```
And when I try 

`./rx -p 0 -d 1 -b 8 -r 4 -f 1024 a021b7cc0b32`

the data start flowing, but I cannot change the script .profile as it get overwritten on reboot...
So, how can I make it work with the right interface?

andrew-grischenko, 2018-04-27:
OK, solved!

The problem was that onboard WiFi got enabled after the upgrade. Following [these steps](https://dephace.com/how-to-disable-onboard-wifi-for-raspberry-pi-3/) I disabled it again and now it works, reading only USB wifi:
```
cd /etc/modprobe.d
nano raspi-blacklist.conf
```
Add in the file and save

```
#wifi
blacklist brcmfmac
blacklist brcmutil
```

`reboot`

And Node.js is also working after that!
BTW, following these steps:

```
curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.8/install.sh | bash
nvm install node 
node -v
```

Thanks! :)

rodizio1, 2018-04-29:
You have upgraded the kernel and the whole distribution and basically screwed everything up by doing that. Kernel patches are gone now and also the configuration changes to the kernel which are required. Plus the stuff that was probably overwritten by the dist-upgrade, startscripts, config files, etc.

I'd start over with a clean image and then check what is it exactly that prevents node.js from being installed and fix only that.

rodizio1, 2018-04-29:
Have you really checked _everything_ after running the dist-upgrade? 

Made sure that cron is not running (there might be be badblocks if cron starts doing it's stuff and use up cpu and access the sdcard while you're flying...) Made sure that no whatever other daemons that might have been added  are running?

Checked that systemd's built-in services (like the journald crap for example) aren't active again?

Chances are, you've screwed it up without realizing it, seeul8er.

seeul8er, 2018-04-29:
Well, of course not. There is no way I could have known about cron and journald messing with the transmission. But (more or less experienced) people are looking into ways of using and modding the image. See it as a hint that prevents major screw ups.

Maybe something like a list inside the wiki, that collects info like this, would be usefull.

rodizio1, 2018-05-02:
Well, cron is part of every linux distribution, it runs periodically and does different things, common linux knowledge. That you don't want any unnessary stuff running for this type of application (where a short 0.5s video stutter or delay can lead to a crashed aircraft ...) especially on the Pi with it's limited ressources and crappy linux multitasking and I/O-scheduling I'd also consider common linux/pi knowledge.

Same goes with doing a dist-upgrade, you don't just do a dist-upgrade, common knowledge.

I have already written some things in the other issue about OpenCV, I'll see that I'll add that info to the wiki.

Question is though, how much will that help, I cannot possibly think of all the things that one can do wrong.
If people want to make changes they need knowledge and experience and do proper analysis and testing, I don't see a way around that.

closed 2018-05-26 by andrew-grischenko