rodizio1/EZ-WifiBroadcast #116: webcam insted of raspicam

bverissimo, 2018-04-29:
Hi,

Thanks alot for all this amazing job :).
I dont have a raspicam  but i want to try to use my hp cam.
In the profile file i didnt understand what to change to make my cam works.
Is it possible to share a profile file with this change?

Many thanks.
cheers

nsimon1, 2018-05-01:
@bverissimo Have a look at what careyer has done using gstreamer: https://github.com/bortek/EZ-WifiBroadcast/issues/71

Search for "AirPi: gst-launch-1.0" within the issue and take a look at their pipeline setup. You should be able to open your webcam using gst-launch-1.0 v4l2src... 

You will change need to change where raspivid is being piped into tx_rawsock in the .profile file once you get your pipeline working/tested.

bverissimo, 2018-05-01:
@nsimon1  Thank you so much to point me in the right direction :).
I will try to make this work because ezwifibroadcast is amazing from what i see in youtube.

:+1:

rodizio1, 2018-05-02:
I have never used a webcam, so cannot really help much. The commented out v4l-ctl line and the line below in the .profile script in the tx_function are left from some tests I made with video4linux and the rasperry cam, maybe you can use or change them for your webcam.

You'll need a little knowledge

closed 2018-11-02 by bortek