rodizio1/EZ-WifiBroadcast #118: Use with Thermal Expert Q1

wstoettinger, 2018-05-16:
Hey!

I'm trying to transmit a thermal image using the Thermal Expert Q1. My test system is up and running (transmitting the raspicam stream works fine). 

My test setup:
two RPis with a Alfa AWUS036NHA each, sufficient power supply, a monitor and ethernet connection for development purposes.

Now the next step would be to read out the stream of the thermal camera. I have successfully initialized the camera and saved images to the disc with a normal raspbian installation (using libpng), however, when I compile and run the program on the wifibraodcast image I keep getting an "illegal instruction" error when initializing the camera.

I'm using the following driver which works fine on raspbian:
https://drive.google.com/open?id=1OcPKFwdkWN9RxaocGh8vI-k9ve-eLDxh

before I install the driver, i run:
```sh
sudo apt-get install -y libudev-dev libusb-dev
```
and before I compile i export the LD_LIBRARY_PATH variable (which i also put into the .profile file):
```sh
LD_LIBRARY_PATH=/usr/local/lib
export LD_LIBRARY_PATH
```

The install script of the driver itself performs some "magic" ... it seems that bitecode and script are stored within one file ... apparently generated with a tool called `makeself` which I don't know ... but it works on raspbian (on the same board and camera) so it shouldn't be a hardware issue.

here is my code `te.cpp` (which is mostly the demo code of the camera documentation):
```c++
#include <iostream>
#include "i3system/i3system_TE.h"

using namespace i3;
using namespace std;

int main()
{	
    // Scan connected TE.
    SCANINFO *pScan = new SCANINFO[MAX_USB_NUM];
    ScanTE(pScan);
    int hnd_dev = -1;
    for(int i = 0; i < MAX_USB_NUM; i++){
        if(pScan[i].bDevCon){
            hnd_dev = i;
            break;
        }
    }
    delete pScan;
	
    // Open Connected Device
    if(hnd_dev != -1){
        cout << "TE found" << endl;
        TE_B *pTE = OpenTE_B(I3_TE_Q1, hnd_dev);
        bool exit = false;
        if(pTE){
            cout << "TE Opened" << endl;
            // Read Flash Data
            if(pTE->ReadFlashData() == 1){
                cout << "Read Flash Data" << endl;
                int width = pTE->GetImageWidth(), height = pTE->GetImageHeight();
                unsigned short *pImgBuf = new unsigned short[width*height];
                for(int i = 0; i < 10; ++i){
                    // Get Image Data
                    if(pTE->RecvImage(pImgBuf)){
                        cout << "Image Received" << endl;
                        // Get Tamperature at (x, y)
                        int x = 100, y = 100;
                        float temp = pTE->CalcTemp(x, y);
                        cout << "Temperature = " << temp << endl;
                    }
                    else{
                        cout << "Image not received" << endl;
                    }
                }
                delete pImgBuf;
            }
            else{
                cout << "Fail to Read Flash Data" << endl;
            }
            // Close Device
            pTE->CloseTE();
            cout << "Close Usb" << endl;
        }
        else{
            cout << "Open Failed" << endl;
        }
    }
    else{
        cout << "Device Not Connected" << endl;
    }
    return 0;
}
```
my makefile looks like this:
```makefile
te:
  g++ te.cpp `pkg-config libusb-1.0 --libs --cflags` `libpng-config --ldflags --cflags` -L/usr/local/lib -li3system_usb_32 -li3system_te_32 -o te
```

the output I get when running the program:
```sh
TE found
TE Opened
Illegal instruction
```

So the error seems to happen within the `ReadFlashData()` function.

I then followed the instructions [here](https://stackoverflow.com/questions/10354147/find-which-assembly-instruction-caused-an-illegal-instruction-error-without-debus).

The output of gdb is:
```sh
Core was generated by `./te'.
Program terminated with signal SIGILL, Illegal instruction.
#0  0x76e0e9ea in i3::Q1_B_Impl::ReadFlashMultiOffset() () from /usr/local/lib/libi3system_te_32.so.2
```

That's how far I got so far and would appreciate some ideas on how to proceed. 

Maybe something a different kernel or kernel setting could fix?

Thanks a lot!

careyer, 2018-05-16:
Hi Wolfgang,

nice camera! Did not know about the Thermal Expert Q1 yet. May I ask: What framerate can it deliver?

I did something similar some time ago with a FlirOne for Android and stumbled upon some challenges as well. You might find some valuable information here: #71 

Here are a few things that come to my mind:
a.) Is it possible to compile the driver on a stock Raspbian and copy over the binary to the EZ-WBC image?
b.) If the driver needs compilation of some kernel modules (the FlirOne Driver needed some additional kernel modules) be aware that you have to download the Soruces of the patched kernel for EZ-WBC from the internet and this most likely appends a "+" sign at the end of the kernel release string. That mixed things up for me and it took me a while to realize this.

careyer, 2018-05-16:
You are welcome.
I use a FlirOne Gen2 (discontinued)... it uses a Lepton3 sensor with 160x120. (Same as the new FlirOne Gen3 Pro). The new FlirOne Gen3 "Standard" (only 80x60) is indeed disappointing especially considering that the Gen3 Standard is the same price as the way better Gen2 (with twice the resolution) was. Mind-boggling.  You did not mention frame the frame rate. FlirOne is only 9Hz for all models due to export restrictions.

bortek, 2018-11-02:
I am trying to clean up the clutter, Can you please close this issue if not needed anymore. Thanks!

closed 2019-01-11 by rodizio1

0mnibus, 2019-05-30:
You can describe in detail how you managed to get the stream from the thermal expert module. I am new to Linux and would like to get an image from my Thermal expert on a raspberry pi. If you can write instructions step by step. I am familiar with the terminal in Linux. You can use my mail.