rodizio1/EZ-WifiBroadcast #124: What's up with these settings? They don't work!

maZZiveAttack, 2018-05-27:
I've gotten video to stream via wifi. I can change the OSD settings at they are present but the telemetry it doesn't work. I've made the changes pointed out in the settings. I am a little confused with the Rx and Tx Serial port settings, where to use serial0 or serial1 other than that I don't know why this doesn't work. 

I can connect my drone to Mission Planner and see where I am. Know the direction the drone is point and know it's location. ez-Wifibroad cast, you know people are going to be using Pixhawk cube so why isn't there a step-by-step set of instructions! This is useless to me becuase i've followed the instructions and nothing, but the video, works. What a joke. Thank you DJI for ocuSync!

careyer, 2018-05-27:
Obviously there is a direct connection between politeness and intelligence. 
Before tinkering with drones you should better learn how to properly interact with people. 
This is a free software project - i.e. you are free to not use it. So go home and shut the door from the inside - and please disconnect from the internet dude.

geofrancis, 2018-05-27:
you clearly lack the intelligence to operate this system, perhaps you should stick with DJI.

rodizio1, 2018-05-28:
With default settings in 1.6RC6, Mavlink downlink works right away. 

Most probable cause is your flightcontrol not giving out a telemetry stream until it is told so by Missionplanner. This is a common pitfall and described in the Wiki, you need to get those SR_ parameters right.

closed 2018-05-28 by rodizio1