rodizio1/EZ-WifiBroadcast #126: `profile` script in master branch wildly different than what's in the 1.6RC6 repo

johnboiles, 2018-06-12:
I noticed last night while making some changes to the `profile` script that the file that's in the master branch in this repo is _very_ different from the file I found on my Pi (with the 1.6RC6 image). It seems that the version on my SD card is newer. It seems to also be the case for some of the source files I've looked at (like rx.c).

Do you recommend developers making changes to this code start with the versions of the files that are in the SD card image? Or is there somewhere else (other repo, branch, tag, etc) we should be looking for the latest versions of the source?

I'm working on a project to pipe 360 video from an Insta 360 Air through EZ-WifiBroadcast and then sending it up in a weather balloon! We've got dual-fisheye video going through EZ-WifiBroadcast working like a charm, just need to work out some kinks.

rodizio1, 2018-06-14:
Sorry, the Repo is not up-to-date. I haven't really got around to update it. The images contain all the latest sources though.

tizianofiorenzani, 2018-07-27:
@johnboiles, I'd like to modify the profile so to get a different ip address. Can you guide me through it? Where do I find the file in the Rpi? Do I need to recompile the repo to get it working?