rodizio1/EZ-WifiBroadcast #127: Camera not Detected with Pi3 compute module

geofrancis, 2018-06-14:
I am trying to get wifibroadcast to work with a pi3 compute module but the camera is not detected, i suspect its because most pi's use camera 1 and not cam 0 or its because the compute module needs some gpio tweaks. i have been at it for days now and have got no further.

https://www.raspberrypi.org/documentation/hardware/computemodule/cmio-camera.md

rodizio1, 2018-06-24:
Hmm, not sure what it could be, I'd try disabling the camera check in the .profile and then see if you can get a picture out of it following the instructions on the page you linked.

geofrancis, 2018-07-04:
i got it working with a hardware.dat file on 1.6 r5 but it wouldnt work on r6.

closed 2018-11-02 by bortek