rodizio1/EZ-WifiBroadcast #129: Pixhawk 2.1 cube not recognised as serial device

geofrancis, 2018-06-19:
I have been trying to connect the pixhawk 2.1 to the pi using USB but it's not recognised as a serial device, it just shows on the list as a device number without a name.

rodizio1, 2018-06-20:
Make it throw an error by setting FREQ=3000 (or whatever) and attach the errorlog.txt you'll find on the sdcard

zenuavsolutions, 2018-06-21:
It is not a good idea to run Pixhawk in any vehicle with USB connected, why not using serial connections?

geofrancis, 2018-06-22:
it shows up fine on a stock raspberry pi image as ttyACM0 but not on wifibroadcast, it just says ACM0 not found

geofrancis, 2018-06-22:
05:58:29 up 0 min,  0 users,  load average: 0.74, 0.16, 0.05

Camera: supported=1 detected=1

2cb05dc69862 Link encap:UNSPEC  HWaddr 2C-B0-5D-C6-98-62-90-0D-00-00-00-00-00-00-00-00  
          UP BROADCAST RUNNING MULTICAST  MTU:2304  Metric:1
          RX packets:129 errors:0 dropped:129 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:31617 (30.8 KiB)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

global
country DE: DFS-UNSET
	(2302 - 2742 @ 40), (N/A, 30), (N/A)
	(4910 - 5835 @ 160), (N/A, 30), (N/A)

global
country DE: DFS-UNSET
	(2302 - 2742 @ 40), (N/A, 30), (N/A)
	(4910 - 5835 @ 160), (N/A, 30), (N/A)

Wiphy phy1
	max # scan SSIDs: 10
	max scan IEs length: 2048 bytes
	max # sched scan SSIDs: 16
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * P2P-client
		 * P2P-GO
		 * P2P-device
	Band 1:
		Capabilities: 0x1020
			HT20
			Static SM Power Save
			RX HT20 SGI
			No RX STBC
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
[... 1666 lines cut ...]
tx_queue_data0_aifs=1
tx_queue_data0_cwmin=3
tx_queue_data0_cwmax=7
tx_queue_data0_burst=0

wmm_ac_bk_cwmin=2
wmm_ac_bk_cwmax=2
wmm_ac_bk_aifs=3
wmm_ac_bk_txop_limit=47
wmm_ac_bk_acm=0
wmm_ac_be_aifs=2
wmm_ac_be_cwmin=2
wmm_ac_be_cwmax=3
wmm_ac_be_txop_limit=47
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=2
wmm_ac_vi_cwmax=3
wmm_ac_vi_txop_limit=47
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0

geofrancis, 2018-06-22:
not sure what was happening, it just started working :S

closed 2018-06-22 by geofrancis