rodizio1/EZ-WifiBroadcast #130: Problem using USB H.264 Cam

tops4u, 2018-06-23:
Hi,

I have finally managed to get GStreamer to fidle the Camera as it should but now I don't get any Video Data on the RX Side. 

Telemetry (Mavlink) works, I can see it on the RX Screen as well as on a ETH Conntected Laptop, but Video stays dark. Checked on RX with the rx Command by sending output to a file, but it is empty. As I understand my Console should get flooded too as long as I don't pipe or redirect the output of the rx command. 

On the TX Side I see that tx_rawsock has been started. If I TEE between GStreamer and tx_rawsock I get Data... Just it looks as it is not being transmitted. 

Any Ideas?

rodizio1, 2018-06-24:
Looks like the tx_rawsock doesn't get any data or doesn't want to transmit for some reason.

Maybe check with the pipeviewer ('pv') tool if you get data, something like:

"gstreamer commandline | pv -i 0.5 -r > /dev/null" to see if data comes out of your gstreamer and then:

"gstreamer commandline | pv -i 0.5 -r | tx_rawsock commandline" 

to see if the data flows to tx_rawsock.

Does tx_rawsock report that it's injecting data, or is it just "sitting" there?

Maybe also check with the top command if it's doing something (i.e. using the cpu).

tops4u, 2018-06-25:
I tried with TEE to a File in /tmp. I could see at least Data there (don't know if it made sense). 
TX_Rawsock does not report anything at all. I thought it just takes whatever comes on STDIN an moves it to the WLAN. 

Just to note I have had this HW Setup running before on the WifiBroadcast Sources here https://github.com/svpcom/wifibroadcast 
I had not sooo super duper results for Distance, Quality and Delay and thought I'd try EZ-Wifibroadcast also as it has this nice OSD integrated. I use Double RALINK Sticks from Aliexpress (https://de.aliexpress.com/item/RaLink-RT3572-2-4GHz-5-0GHz-802-11n-a-Dual-Band-300Mbps-Wireless-USB-WiFi-Adapter/32861123628.html?spm=a2g0x.10010108.1000016.1.3a825b7bCHCiV3)
Tried to run them on 2.4 first, but the Futaba TX (S-FHSS) killed the signal.

rodizio1, 2018-06-25:
The other solutions don't work as good as EZ-Wifibroadcast, there have been a lot of improvements, on the wifi driver side, kernel side, and also with the tx and rx programs, see the Release Notes for more info.

Yes, tx_rawsoxk takes whatever it gets in stdin and sends it out. Could it be, that you run it with wrong commandline parameters? It doesn't really check for wrong combinations.

You can check the parameters by using an unmodified EZ-Wifibroadcast image and then running 'ps ax' to see what the parameters are.

tops4u, 2018-06-25:
Hi I succeed today in getting it working. As somebody wrote in another Issue I had to modify the Source to take /root/video which is a named Pipe as Input and it worked. Before I got Invalid File Descriptor when trying via cat of the very same file. 

However Video Performance is very poor, but I think this is because of limitations of my USB Camera which has no means to change Keyframe Rate or Bitrate... I have ordered a Raspi-Cam for testing. Nevertheless I would very much like to use a Composite -> USB Stick with this solution in order to replace the analog FPV Transmitters. As I still have an analog FLIR TAU I would be happy if you know if somebody has done something like this before.

closed 2018-08-16 by tops4u