rodizio1/EZ-WifiBroadcast #132: Errors when applying patches

terial, 2018-06-29:
When trying to apply the patches from EZ-WifiBroadcast/kernel/ i get errors. I am using the Raspberry 4.9.35 kernel sources from the link, https://github.com/raspberrypi/linux/archive/be2540e540f5442d7b372208787fb64100af0c54.tar.gz.  I tried all the patches except linux-4.9.28-dwc-otg.patch since I am using 4.9.35, linux-4.9.28-brcmfmac-intwifi.patch and linux-4.9.28-brcmfmac-tos.patch as I dont need to change the internal wifi stuff.

Here is what I get when applying the patches...

**linux-4.14.32-ath-regd.patch**
admin@FCC_RSP01:~/linux $ sudo patch -p1 < /home/admin/patch/linux-4.14.32-ath-regd.patch
patching file drivers/net/wireless/ath/regd.c
Hunk #6 FAILED at 250.
1 out of 6 hunks FAILED -- saving rejects to file drivers/net/wireless/ath/regd.c.rej

**linux-4.14.32-rt2800usb-misc.patch**
admin@FCC_RSP01:~/linux $ sudo patch -p1 < /home/admin/patch/linux-4.14.32-rt2800usb-misc.patch
patching file drivers/net/wireless/ralink/rt2x00/rt2x00lib.h
Hunk #1 FAILED at 29.
1 out of 1 hunk FAILED -- saving rejects to file drivers/net/wireless/ralink/rt2x00/rt2x00lib.h.rej

**linux-4.9.28-ath-regd.patch (seems like a patch is already applied?)**
admin@FCC_RSP01:~/linux $ sudo patch -p1 < /home/admin/patch/linux-4.9.28-ath-regd.patch
patching file drivers/net/wireless/ath/regd.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
6 out of 6 hunks ignored -- saving rejects to file drivers/net/wireless/ath/regd.c.rej

**linux-4.9.28-ath9k_htc-misc.patch (seems to work fine)**
admin@FCC_RSP01:~/linux $ sudo patch -p1 < /home/admin/patch/linux-4.9.28-ath9k_htc-misc.patch
patching file drivers/net/wireless/ath/ath9k/common-init.c
patching file drivers/net/wireless/ath/ath9k/eeprom_4k.c
patching file drivers/net/wireless/ath/ath9k/eeprom_9287.c
patching file drivers/net/wireless/ath/ath9k/hif_usb.h
patching file drivers/net/wireless/ath/ath9k/hw.c
patching file drivers/net/wireless/ath/ath9k/hw.h
patching file drivers/net/wireless/ath/ath9k/mac.c

**linux-4.9.28-crda.patch (seems to work fine)**
admin@FCC_RSP01:~/linux $ sudo patch -p1 < /home/admin/patch/linux-4.9.28-crda.patch
patching file net/wireless/db.txt
patching file net/wireless/reg.c

**linux-4.9.28-rt2800usb-misc.patch (seems to work fine)**
admin@FCC_RSP01:~/linux $ sudo patch -p1 < /home/admin/patch/linux-4.9.28-rt2800usb-misc.patch
patching file drivers/net/wireless/ralink/rt2x00/rt2800usb.c
patching file drivers/net/wireless/ralink/rt2x00/rt2x00dev.c
patching file drivers/net/wireless/ralink/rt2x00/rt2x00lib.h
patching file drivers/net/wireless/ralink/rt2x00/rt2x00link.c

**linux-4.9.28-rt2800usb-txpower.patch (seems to work fine)**
admin@FCC_RSP01:~/linux $ sudo patch -p1 < /home/admin/patch/linux-4.9.28-rt2800usb-txpower.patch
patching file drivers/net/wireless/ralink/rt2x00/rt2800lib.c
patching file drivers/net/wireless/ralink/rt2x00/rt2800lib.h
patching file drivers/net/wireless/ralink/rt2x00/rt2800usb.c
Hunk #2 succeeded at 871 (offset 2 lines).

The 3 that down work are
linux-4.14.32-ath-regd.patch
linux-4.14.32-rt2800usb-misc.patch
linux-4.9.28-ath-regd.patch

Any suggestions or tips to troubleshoot?

rodizio1, 2018-06-29:
You are applying the 4.14 patches to a 4.9 kernel ...

The 'already applied' message is exactly what it says, you already applied those changes.

terial, 2018-07-01:
Thanks for the reply, for linux-4.9.28-ath-regd.patch that was the first time I applied it and got that error... Do the 4.14 patches no longer apply (as in the newer 4.9 kernel doesnt need them?) or is the repo just not current with what you have made? I know you cant keep it all current all the time, just curious. Thanks again for all your work!

rodizio1, 2018-07-01:
4.14 kernel is newer than 4.9, not the other way around.

4.14 patches are for 4.14 kernels, for people who want to experiment with newer 4.14 kernels.

If you use the 4.9 kernel mentioned in the readme and only apply the linux-4.9... patches, it works just fine.

bortek, 2018-11-02:
closing as this is not relevant at the moment

closed 2018-11-02 by bortek