rodizio1/EZ-WifiBroadcast #135: Video switching patch from user htcohio

rodizio1, 2018-06-29:
https://www.rcgroups.com/forums/showpost.php?p=39867908&postcount=4932

https://drive.google.com/file/d/1vHvJlccpbVlma0mXGT332akWsoPuuQH3/view

careyer, 2018-06-30:
This is awesome! However, this solution occupies one of the already very limited 8 MavLink channels. It would make a whole lot of sense to expand the 8 proportiol channels (MavLink) with a number of additional switching channels (e.g CH9-12 maybe even CH9-16) which can either control GPIOs (On-Off) or can be interpreted in the WBC-loop to trigger additions in the code (e.g. switch between different OSD screens, trigger a python script, switch multiple cameras etc...) The benefit of this would be unlimited. Right now the only limitation of EZ-WBC is the 8Ch limitation which results from the MavLink override channels mimik. It would be awesome if Rodizio could overcome this by sending more channels to the air which do not nessesary need to be mavlink channels.

htcohio, 2018-07-01:
User1321 is the one who actually came up with the code to get this working as is.  I agree, the possibilities would be endless if this could be figured out.  You would be able to remotely trigger various video or other functions from the ground remotely including data rate and FEC settings.  Could require a reboot while in the air but if you're pushing distance at data rate 3 you could switch to data rate 2 for instance.

rodizio1, 2018-07-02:
Changing wifi bitrate on-the-fly would also be possible, with Atheros cards this would require changes to the kernel drivers and maybe also Atheros firmware, as well as to raspivid.

With Ralink and Realtek it's easier, they already support to set wifibitrate on a per-packet basis.

Another easier to implement option would be changing the bitrate with iw/iwconfig and then re-starting raspivid, but that would probably take about 1-3 seconds which sucks in my opinion.

matiaspl, 2018-07-03:
For the record - gstreamer's gst-rpicamsrc
<https://github.com/thaytan/gst-rpicamsrc> allows live bitrate switching

htcohio, 2018-07-03:
The primary advantage of changing Wi-Fi bitrate on the Fly is that you can receive video at lower and lower noise floor levels (i.e.) greater range.  That's my understanding at least...

JamesCordell, 2018-07-09:
The raspivid does support variable bitrate but I think it adapts based on the image complexity and may not be "commanded to switch on an adhoc basis". I'll look at this.

closed 2018-11-02 by bortek