rodizio1/EZ-WifiBroadcast #138: RC passyhrough

bazyl78, 2018-07-16:
Can you try to do rc passthrough? For example ppm input on ground pi and ppm output on air unit?. That would be great to use this as a module for rc transmitter and have all in one rc control solution like OpenSolo.

rodizio1, 2018-07-17:
Sorry it's not possible to output PPM in a 100% reliable way on the air Pi. 

You can use a PPM to USB-Joystick adapter on the ground Pi though.

On the air Pi, you can chose between several different protocols like Mavlink, IBUS, SUMD,  XBUS (all digital and checksummed, thus 100% safe and reliable), most flight controls support at least on of these.

careyer, 2018-10-18:
Sorry PWM / PPM / CPPM ... is not possible on RaspberryPi, see: https://github.com/bortek/EZ-WifiBroadcast/wiki/RC-:-regular-RC-protocols

closed 2018-10-18 by careyer