rodizio1/EZ-WifiBroadcast #140: Joystick needs to be connected after RX startup

rodizio1, 2018-07-31:
As reported by Trebor Link:
One bug is if the joystick is plugged in before booting the RX does not see any wifi adapters ?
Have to have the Rx running, plug in the joysick then start the TX and FC.

dudman, 2018-08-07:
not directly connected, but wonder what the latency is like with a joystick over wifi, is it still flighable ? and fun for fast wings ?

pilotnbr1, 2018-08-25:
Latency is extremely low for rc. The latency of the video (camera latency, compression, transmission, decompression, display) should not be confused with the rc latency. As for a hard number I cannot tell you.

bortek, 2018-11-02:
I am trying to clean up the clutter @pilotnbr1 , do you know if this is still relevant or can we close it ?