rodizio1/EZ-WifiBroadcast #144: 3 Atheros cards + 1, 5.8 card result in freeze up during boot.

htcohio, 2018-08-15:
Id like to bring up one more item related to an issue running four wifi cards (3 Atheros for RX + 1 non atheros for the 5. 8 hotspot).

Three-way diversity on the ground with atheros cards work perfectly fine by themselves however, if I connect fourth card for the hotspot and try to boot it freezes up after it loads the first or second card.

Are there any possible Immediate Solutions to this? is there any way to connect 2 RX pis to eliminate this problem? this would also provide a few extra USB ports for other things such as keyboards and USB app functionality?

Thank you.

bortek, 2018-11-02:
I am trying to clean up the clutter and will close this one as the code is under major overhaul. Retest with new image when it's available.

rodizio1, 2018-11-07:
Unfortunately, it only works up to 3 Atheros or 2 Atheros and 2 Ralink, doesn't matter if the cards are used for hotspot or wbc. 

Another solution for 3x Atheros plus 5GHz hotspot would be the new Pi3B+ with it's new 5Ghz capable wifi chip.

htcohio, 2018-11-07:
Hey, yes the internal 5G on B+ Works flawlessly.

careyer, 2018-11-07:
It also works with 3x Atheros and the build-in 2,4Ghz Pi3B+ Wifi. So far I could not observe that the low-power internal Wifi of the Pi3B+ interferes with the WBC link or has noticeable effect on the packet loss rate.

rodizio1, 2019-01-11:
closed since it can't really be fixed, would require re-writing the whole Atheros drivers and parts of the Linux Kernel and Raspberry Firmware.

closed 2019-01-11 by rodizio1

zipray, 2019-01-21:
Not clear why "ifconfig $1 up" twice in prepare_nic?
https://github.com/rodizio1/EZ-WifiBroadcast/blob/2db9e268a69f86580a1a370678fee67544e836b1/root/.profile.save#L249

https://github.com/rodizio1/EZ-WifiBroadcast/blob/2db9e268a69f86580a1a370678fee67544e836b1/root/.profile.save#L304