rodizio1/EZ-WifiBroadcast #145: Add Optimising power consumption page to the wiki

gba-gbob, 2018-08-26:
Hi, did some work on power consumption, hopefully this can be useful to others.

`###Optimising power consumption

In the world of micro fpv sensible power consumption is important. Below measurements should give you rough idea on what savings can be made.  

| consumption ma@voltage | change ma |config change            |notes    |
| :---------------------:|:---------:|:-----------------------:|:-------:|
| 400-430@5v             | 0         | none                    |         |
| 490-530@3.8            | 0         | none                    | keeps rebooting not usable |
| 400-430@5v             | 0         | force_turbo=0           |         |
| 360-370@5v             | -50ma     | arm_freq=600            | 10% cpu idle VIDEO_BITRATE=auto, 48% cpu idle VIDEO_BITRATE=3000 |
| 300-320@5v             | -50ma     | VIDEO_BITRATE=3000      |         |
| 290-310@5v             | -10ma     | FPS=30                  |         |
| 310-350@5v             | +30ma     | DATARATE=2              |         |
| 300-330@5v             | +15ma     | DATARATE=3              |         |
| 290-310@5v             | 0         | DATARATE=4              |         |
| 260-290@5v             | -25ma     | /usr/bin/tvservice -o   | switching of hdmi output  |
| 260-290@5v             | 0         | over_voltage            | values -4 to 6 no effect  |
| 260-290@5v             | 0         | over_voltage_sdram      | values -4 to 6 no effect  |
| 260-290@5v             | 0         | txpower_ralink          | values  -5 to 5 no effect |

#### After optimization
* 260-290@5v (100-110ma wifi card)  
* 360-390@3.8v (130-150ma wifi card)  

#### Notes
* Setup: rasp cam v2, M08-50 RT5370N wifi module  
* Lipo battery wired directly into header pins 2->3.8 VDC, 6->GND  
* Very likely TX power adjustment had no effect on this particular setup  
* Setting over_voltage less than -4 causes reboots  
* Reducing cpu arm_freq below 600 has very minimal effect power consumption < 10ma
* Measured with Vici VC99 multimeter  

##### TODO
* Find wifi card obeying tx power settings hopefully allowing sub 200ma@5v setup.   `

pilotnbr1, 2018-10-12:
Nice contribution!

careyer, 2018-10-18:
Thank you very much... merged into Wiki: https://github.com/bortek/EZ-WifiBroadcast/wiki/Expert-:-Optimizing-Power-Consumption

closed 2018-10-18 by careyer