rodizio1/EZ-WifiBroadcast #148: Upgrade to newer Distribution

tops4u, 2018-08-31:
I wanted to use a standard Composite to USB Stick with GStreamer, after some hours of trying I noticed that EZ-Wifibroadcast is using a rather old Linux distribution where the current Version of GStreamer is not available as DPKG. I was not yet successfull in compiling it for the current Distribution and the newest GStreamer Package available will not run with the USB Dongle, while newer GStreamer Binaries work. 

Any intention of upgrading? If so I'll try to use a maiden installation of a current distribution and migrate the Wifibroadcast modifications to it.

Yes21, 2018-08-31:
You could be interested by this project : [wifibroadcast-image-builder](https://github.com/RespawnDespair/wifibroadcast-image-builder).
It can build ez-wbc 1.6rc6 on a newer raspbian version (2018-06-27) with a 4.14 kernel, and support for rpi3b+.

roque-canales, 2018-09-01:
Yes21, I dont know how to use wbc imag Builder.

Please tell me how to for I test last kernel with 8812au driver.

Yes21, 2018-09-04:
You can find [here](https://drive.google.com/open?id=14kkL5wKGUA06eq9J9Ylt5AkKRveY6ruu) a working image of ez-wbc 1.6rc6 based on a 4.14 kernel + patches. This image should work with Pi3b+.

The image as been created thanks to [wifibroadcast-image-builder](https://github.com/RespawnDespair/wifibroadcast-image-builder).

kwatkins, 2018-09-05:
@Yes21 thanks for doing that, also, the image doesn't have the ez-wifi config files in the boot partition... is it not supposed to, or, as is often the case, am I missing something?

tops4u, 2018-09-05:
I noticed that too. My Mac also complained about an inconsistency of the Boot Partition. I'm about to build my own image. I'll probably post it here. 

@Yes21 Thanks for the Hint with the Builder scripts.

kwatkins, 2018-09-05:
Ahh... Pi3+, mac.

Yes21, 2018-09-06:
> You can find here a working image of ez-wbc 1.6rc6 based on a 4.14 kernel + patches. This image should work with Pi3b+.

I'm really sorry, but my image doesn't work. I've made an error in testing it, and I can know confirm your observations.
I will create new issues, and try to help to resolve them.

kwatkins, 2018-09-06:
@yes21 no worries, big thanks for doing this

Yes21, 2018-11-02:
Will be done with 1.6.

closed 2018-11-02 by Yes21

rodizio1, 2019-01-11:
Won't be implemented in v1.6 as this is a very large change that has the potential to break things and we are currently already in the release candidate (i.e. only small changes and bugfixes to be made) stage.