rodizio1/EZ-WifiBroadcast #149: CPU temp and usage does not get forwarded for gnd pi

Consti10, 2018-09-08:
Hello Rodizo,
It seems like cpuload_gnd and temp_gnd does not get forwarded with the ez-wb forward struct.
Can you have a look at it ?

Greetings,
Constantin

careyer, 2018-11-02:
@bortek : Sorry I am missing the context to understand Constis question correctly. @Consti10 - can you give us a bit more context and details on your above question? We would be happy to address this problem if it still persists.

Consti10, 2018-11-03:
Version 1.6rc6 on both rpi's.
FPV-VR connected via usb tethering.
These are the values of the ez-wifibroadcast-forward-t struct:
[image]

Note that cpuload_air and temp_air are transmitted,
but cpuload_ground and temp_ground are not.

Hope this helps !

careyer, 2018-11-06:
@Consti10 thank you for providing additional information. @rodizio1: Can you confirm that cpuload_gnd and temp_gnd does not get forwarded (or are empty) with the ez-wb forward struct? We could then apply the "bug" label

Consti10, 2019-01-05:
Hello everyone,

So I decided to go trough the EZ-WB code myself searching for the bug, and I noticed something strange:
The only code related to the ez-wb status data forwarding is this file: 
[https://github.com/rodizio1/EZ-WifiBroadcast/blob/26514f7d6d7f7ec42e3d16e98a4b164f6bd4870e/wifibroadcast/rssi_forward.c](url)

but it seems to be the old one, forwarding the best rssi value. This makes me think: Is the code here on this branch even version 1.6 rc6 ? I mostly had to treat ez-wb as a big black working! box  but since rodizio seems to be fully occupied by other means we might have to figure this one out ourselves

rodizio1, 2019-01-11:
Sorry for not replying earlier Consti. I'm back working on it. Currently, these values don't get forwarded, because it's just not implemented.

The Code on Github is not up-to-date, look into the 1.6.RC6 image for latest source. I'll see that I clean up this Github mess in the near future.

closed 2021-04-23 by Consti10