rodizio1/EZ-WifiBroadcast #151: GPIO pins used for custom profiles

wowi69, 2018-09-23:
page: https://github.com/bortek/EZ-WifiBroadcast/wiki/How-to's
diagram: https://raw.githubusercontent.com/bortek/EZ-WifiBroadcast/master/wiki-content/Dipswitch.JPG

it names GPIO pins 28,7,24,23

shouldnt this read 1,7,24,23 ?
(wifibroadcast-misc/gpio-config.py)

htcohio, 2018-10-19:
> Or rather look here which is a file I extracated from latest 1.6RC6. There
> we have complet elist of config files. According to ti it should Be GPIO
> 1,7,24,23
>
> input_state0 = GPIO.input(1)
> input_state1 = GPIO.input(7)
> input_state2 = GPIO.input(24)
> input_state3 = GPIO.input(23)
>
>
> https://github.com/bortek/EZ-WifiBroadcast/blob/1.6RC6/root/wifibroadcast_misc/gpio-config.py#L14
>
> Keep in mind these are GPIO pint not physical pins on Pi.
>
> Moreover what is this thing for where are the dip switches, is it on
> flight controller, then which flight conrtoller? I am looking at my
> Pixhawk1 and that does not have ani dip switches. The docs is pretty
> unclear about it.

I'm trying to figure that out myself.  Soon this may not even be relevant
though because pre-saved profiles and simultaneous synchronized setting
changes will be handled through an app eliminating the need to edit text
files.

We already have code that serves this purpose but we are quite at that step
in our roadmap yet ))

This gpio (dipswitch) function could still potentially be used as more of a
physical button to "reset stock wfb defaults" in the event air/ground unit
do not have the same settings.

With that part planned in the very near future, do you think that it would
still be worth looking into and documenting?

careyer, 2018-10-19:
@bortek  : Thanks for diggging into to. Yes you have of course to search for the gpio-config.py in the latest 1.6RC6 Build. I did so as well. 
You mixed things up a bit. The Dip-Switches are supposed to be added to the RaspberryPi header and allow accroding to the Dip-Configuration to load 16 different EZ-Wifibroadcast configurations (from wifibreadcast_1.txt .... wifibreadcast_16.txt)

However what buffels me
- In the above depicted PI-Pinheader I cannot see and GPIO marked GPIO1, can you?
- There is also no GPIO called GPIO28 

So I wonder: Which pins is Rodizio refering to?

Apprt from that:
@htcohio : T believe this functionality is still usefull and we should keep it. However NOBODY! needs 16 different configurations. These Dip switches are occuping precious GPIO pins. I belive this this be reduces to lets say 4 configurations --> 2 Dip Switches

RespawnDespair, 2018-10-19:
GPIO 1 is reserved for the SD card, this cannot be used. In the original 8 config code i see 7, 23, 24.
These are off course available on pin 16, 18 and 26.
I agree we should limit the amount of configs to 4 by using only 2 pins (23, 24 would be convenient i think).

careyer, 2018-10-19:
Good... so there is definitely something pretty screwed up here.
@Yes21 : May I asky you to add this is a task on the roadmap to address/fix this problem. Good that @RespawnDespair  agrees to strip it down to only 2 Pins (4 options). This will free up some GPIOs for better stuff.

careyer, 2018-10-19:
Personally I don't use them either... (same as for @pilotnbr1) ... however when I remember correctly there was a huge demand back then from people who were experimenting with different FEC settings on the field to find the sweetspot and tuning parameters. We can either ditch that feature alltogether or repair it and reduce it down to 2 dip switches. - just my 2ct.

Whatever the case we should correct the documentation as it simply is wrong and confusing.

Yes21, 2018-10-19:
**@careyer  @bortek @RespawnDespair** 
If it's easy to fix it now using only 2 pins (23, 24), I think it should do. 
Is it as easy as deleting to lines in gpio-config.py ?
If yes, I propose that **@RespawnDespair** could correct the code in his repo.
In this case, I will mark this as a bugbix from 1.6, and you could update the wiki right now.

What are your opinions ?

RespawnDespair, 2018-10-19:
@Yes21 I left that part out completely when making the image builder. Once i have the basics working again i will reintroduce this, but with two pins as suggested. I'll make a bug on my board for this.

Yes21, 2018-10-19:
**@RespawnDespair** Ok, I will stick this issue with 1.6 project in this repo

pilotnbr1, 2018-12-02:
@rodizio1 care to weigh in on this issue?

rodizio1, 2018-12-02:
Yes, there was some mix up with the numbers (I confused numbers as there are three different naming schemes for those pins), search the rcgroups forum thread for more info.

>However NOBODY! needs 16 different configurations. These Dip switches are occuping precious GPIO pins. 

Oh, yeah, careyer as I know him:  Mainly pursuing to get "his" functionality into the image (i.e. freeing up those gpios for his intended use) and writing with a staggering confidence about things he doesn't know much about and of course didn't bother to search for or ask before making such bold statements.

4 GPIO pins have been explicitly requested by a user, that's why I put it in. Also I know quite some people are using this functionality (maybe not with 16 different profiles, but surely with more than 4). The reasoning for needing more switches is simple: As soon as you need to change more than one thing, you'll use up those 16 profiles rather quickly. Think of  different cam profiles, maybe one for bright daylight, one for evening. Combine that with different frequencies and your combinations are already cut in half (four different wifi channels plus night/day config would already eat up 8 combinations).

https://github.com/rodizio1/EZ-WifiBroadcast/issues/60

>I agree we should limit the amount of configs to 4 by using only 2 pins (23, 24 would be convenient i think).

Can you explain why you want to break already existing functionality? Currently, the only thing that I can think of that would justifying breaking existing functionality is a new feature with high demand, e.g. audio, for that we might need those pins for an I2S microphone.

Of course, as soon as we have a better way of configuring things, we don't need them anymore and can do different things with them.

rodizio1, 2019-01-11:
> it names GPIO pins 28,7,24,23
> 
> shouldnt this read 1,7,24,23 ?
> (wifibroadcast-misc/gpio-config.py)

Yes, quoting myself from here: https://www.rcgroups.com/forums/showpost.php?p=39256689&postcount=3942

> I just realize the table in the wiki is misleading, it uses inconsistent numbers, sorry for the confusion.
> The new pin should be named "1" in the table, not 28, your picture is correct.

Image in the Wiki is fixed now.

closed 2019-01-11 by rodizio1

mathiasseyfert, 2019-03-01:
Thank you for still including the switch-feature. If EZ-WBC is used on other systems than UAV, it will be very helpful to change frequencies on scene during a live broadcast. 
I am building two systems for mobile cameras at events (shoulder mounted camcorder and cable cam) where it will be eays to tell the camera operator to change to another frequency if needed.