rodizio1/EZ-WifiBroadcast #156: wiringPi doesn't work with kernel 4.9.*

cyrilknops, 2018-10-12:
Trying to add a voltage sensor to the RX pi using the SPI bus with the wiringPi library so that you can see the battery voltage of the receiving Pi when on location.  
 But i get the "unable to determine hardware version, I see: hardware: BCM5835" "expecting DCM2708 or BCM2709" i read somewhere that library doesn't work with the new the 4.9.* kernel only 4.4.*   
I'm not that great at kernels so has anyone have a fix for this?

careyer, 2018-10-12:
@cyrilknops : Good to see you here! Thank you very much for your issue report. I am not at all a developer but I hope that some of my collegues can take care of that question? @user1321 @RespawnDespair  maybe? Cyril wants to integrate a voltage sensor for the GroundPi when powered via battery. Since Raspberry does not offer AanalogDigital Pins he has to use a voltage sensor (I2C) but I2C does not seem to work. Maybe you can help here?

cyrilknops, 2018-10-12:
@careyer Thanks good to be here, nice to se some life back into the project, together with some local flying buddy's we are working on some new features, so you will here from me often.

bortek, 2018-11-02:
I am not sure if I2C bus is working or at all enabled in the Ezwfb image. This will require testing/development to get this working I guess. Not sure we have time for this. @cyrilknops Perhaps you can lab on your own and if you get it working you are welcome to do a PR.