rodizio1/EZ-WifiBroadcast #161: Consistency in Project

pilotnbr1, 2018-10-17:
While doing some work on the wiki and going looking at text within code it is apparent that several different names are being used to refer to the pi. To make it easier for a user I propose we try to all be consistent in our terms, particularly in the Wiki. I believe the use of “Airpi” and “Groundpi” is the most clear. Thoughts?

RespawnDespair, 2018-10-17:
TX/RX is used a lot, but this is almost meaningless to most new users.
Air and ground make the subject a lot clearer. If we want to do it right, we would have to modify the sources as well in an upcoming iteration.

careyer, 2018-10-17:
Thank you @pilotnbr1 for bringing this up! Yes it was me how added those terms "AirPi" and "GroundPi" to the Wiki. I introduced them about 2years ago and they are also commonly used in the RCgroups forum.

The idea behind it is the following: TX/RX do not cut it anymore. Since we are dealing with a bi-directional link both Aircraft and Groundstation both transmit and receive. Depending on whether you are talking in a video, telemetry or RC context TX and RX can mean anything and confuse things quite a lot. So speaking of "AirPi" and "GroundPi" solves this problem entirely.

pilotnbr1, 2018-10-17:
Yes agree 100 percent. Really confusing now with RC and rx/tx... I think we have an opportunity here to “get it right”.

careyer, 2018-10-17:
BTW: I believe the awesome FAQ should not be the last thing on that menu... I moved it to a more prominent place... I renamed the "Introduction"-Chapter to "General" and put it there

bortek, 2018-10-17:
👍

bortek, 2018-10-18:
I think we agreed on terminology for ground and air pi, if yes can you please close the issue?

closed 2018-10-18 by pilotnbr1