rodizio1/EZ-WifiBroadcast #162: Home Page Improvement

pilotnbr1, 2018-10-17:
More of an observation.. Looking at the insights for this repository and traffic, most unique visitors are going to the home page. I believe that page needs to be significantly improved in order to attract users.
-move all older links to past images to the history page
-make page more informational as to what wbc is
-perhaps some multimedia content- @htcohio seemed interested in video
-basically this is where the project will be “sold”

I can begin work work on this if there is support. I just don’t want to unilaterally change such a significant page.

[image]

bortek, 2018-10-17:
I think the firsts page is the one that comes from the readme.md file. We have all the content on wiki. Is it possible to have a landing page a wiki main page? That way we would have one landing page instead of readme.md

careyer, 2018-10-17:
@pilotnbr1 : This is an awesome idea! We could also use maybe khomyakk's 32km world record flight and reedit the video to add some "teaser" character to it... This way we would have both a teaser and and awseome tech demo of what this system is capable of. I know that khomyakk is willing to share the video because he sent me the full 6GB file in the past already. Of course we need to ask him again if it is okay with him

careyer, 2018-10-17:
@bortek : I could not find any informaton on how to redirekt the main landing page of a repository to its wiki.... However I have asked a question how one could achieve this here: https://github.com/contact

careyer, 2018-10-17:
Just in case I have reserved the account for future use:
[img]http://www.ip-networx.de/content/GitHub/EZ-WifiBroadcast_YT_channel.JPG[/img]

We should agree on a code of conduct using it.
- Credentials should be shared at least among two users 
- Any upload should be approved by a plenum 
- Monetization should always be truned off on all videos

RespawnDespair, 2018-10-17:
If we were to have an official logo, we could also set it as a watermark on the channel...

bortek, 2018-10-17:
Googled for half-an-hour and tested different suggestion on linking README.md to Wiki but it seems not possible. 

Which means that we have to create a landing page in the README.md thats how most of the projects are done. The pros of this is that README.md get version controlled and can 'be updated with every release.

pilotnbr1, 2018-10-17:
@bortek when clicking on the “home” link from the insights traffic page, it goes here
https://github.com/bortek/EZ-WifiBroadcast/wiki

So interestingly enough the readme.md page which is 
https://github.com/bortek/EZ-WifiBroadcast is the number 2 page with unique visitors

The analytics don’t make it very clear how they are getting there. Google being the number 1 source of traffic perhaps points to the “home” page as most relevant?!

Both pages could definitely be improved but I suggest most “advertising” go to the page attracting the most unique visitors- https://github.com/bortek/EZ-WifiBroadcast/wiki

@careyer yes good idea with a YouTube channel!

richardbmx, 2018-10-18:
@pilotnbr1  Hi, my name is Richard. I'm a graphic designer and I love making contributions for open source projects. I see that they want a new logo for your application. If you wish, I can do it for you without any problem ... I can present some modern ideas that would improve the appearance and image of your project

htcohio, 2018-10-18:
@richardbmx,

My brother-in-law is working on a couple logos in his free time as we speak.  They just had a baby a month ago so his hands are kind of full but he said he'd have a few more options over to me this week.  I'll post a couple of the samples on rcgroups thread for you to take a look at but, it would be great if a couple people could come up with ideas.

bortek, 2018-10-18:
@richardbmx Yes please do so, come up with some suggestion. We can can then vote and select whichever we thinks suits best to the project. Thank!

careyer, 2018-10-18:
@richardbmx, thank you very much for offering your help. I can't wait with what you will come up with. :-)

RespawnDespair, 2018-10-18:
@richardbmx , i have scrolled through your portfolio, some great logo designs there! @htcohio 's brother-in-law is also going to provide some designs, but i'm very interested in what you would come up with.

When we progress further we would also need some illustrations done for the guides and howto's. I would love to have one consistent visual style for those as well. Perhaps we could look into that when the basics are done in regards to the 1.6 generated image.

RespawnDespair, 2018-10-18:
@bortek looks good to me... I once wanted to make a 'real' manual as i've seen in other projects. So i created and open-source gitbook account.
[https://ez-wifibroadcast.gitbook.io/project/](https://ez-wifibroadcast.gitbook.io/project/)

I think we should stick to the wiki for now, but at least the url is claimed within gitbook 😄

careyer, 2018-10-18:
> @richardbmx , i have scrolled through your portfolio, some great logo designs there! @htcohio 's brother-in-law is also going to provide some designs, but i'm very interested in what you would come up with.

@richardbmx I did so as well... and I really like the light and modern style of the logos I see there! Awesome work! I'm excited.

>When we progress further we would also need some illustrations done for the guides and howto's. I >would love to have one consistent visual style for those as well.

+1 on this one as well.. I very much enjoy for example the very consistent visual appeal of the Ardupilot manual http://ardupilot.org/copter/docs/frame-type-configuration.html / I really enjoy browsing those sites.

pilotnbr1, 2018-10-18:
@richardbmx @htcohio Yes more logo options would be better than less! Personally I had something in mind, that captures the essence of the project- like a camera lenses, WiFi symbol, quadcopter, etc. But open minded and curious to see what is presented.

It is a little strange to go through this wiki and not see any demo videos, pics of osd, or much multimedia- funny for a project that is about hd video, quality, and visual appeal lol

@bortek yes your proposal sounds good. That is logical and seems in line with standard operating practice on github.

All good ideas here!

pilotnbr1, 2018-10-18:
@bortek I have one addition to the readme which is license info, recognition of contributed/borrowed code, and perhaps some sort of disclaimer for legal. This will also make us more compliant with the rest of github.

Additionally github likes to see some sort of code of conduct. I just suggest some boiler plat standard template but most other large projects have this and seems to be good practice.

bortek, 2018-10-18:
Understood about the limitations of landing page. Let's not split attention too much for now, we can take a look at alternative pages on a later stage.

I knew a question about licensing would sooner or later would pop up, so here ot came 😁.  I have an idea and will add it to Readme.

htcohio, 2018-10-18:
Yeah, @richardbmx if you were to handle the logo I think it would be
better.  we would have more control over the graphics and the turnaround
time would probably be quicker for any add-ons or future changes.

Portfolio looks awesome!

richardbmx, 2018-10-18:
perfect I will start working on this

careyer, 2018-10-18:
@richardbmx : you might want to look into this new thread: #164

richardbmx, 2018-10-19:
I would like to present my proposal based on all your ideas I have made this logo that expresses the camera and the Wi-Fi connection very well in addition to this it can be adapted to any background color and size without losing quality. If you want any change you should only ask for it.

[image]

bortek, 2018-10-19:
I have update the landing page text. Can someone take a look and give me a feedback on the text content, don't be shy giving the critic. 

https://github.com/bortek/EZ-WifiBroadcast/blob/develop/README.md

pilotnbr1, 2018-10-19:
@bortek fixed a couple typos. FYI “infamous” has negative connotations (famous in a bad way) don’t think that’s what you meant with Befinitiv:)

I am also not sure about describing wbc as a unidirectional link. With rc and some telemetry wbc is bidirectional- so maybe not describing the capabilities of wbc completely in that regard. Otherwise really liked it! Big improvement over the previous version! Thanks for the hard work!

bortek, 2018-10-19:
good findings, I have changed uni to bi and in*famous have been already corrected by you. Will continue to update it in the coming days.

careyer, 2018-10-19:
added a missing "on" :-) Otherwise pretty much perfect!

careyer, 2018-10-29:
@bortek : You you confirm that you have received the 31km flight video on your NextCloud?

bortek, 2018-10-29:
Oh yeah I can see 117 files in your folder. That took it's time. WIll fetch it now down on my PC then what we can do with it :)

rodizio1, 2019-01-11:
Thanks a lot for your help guys.

closed 2019-01-11 by rodizio1