rodizio1/EZ-WifiBroadcast #164: WBC Logo

pilotnbr1, 2018-10-18:
I opened this in an effort to keep things organized and focused. I suggest we discuss submit logos here. I threw this together in 5 minutes but it was just some concepts I had in my head-
[image]

htcohio, 2018-10-18:
@luke @all,

Here are some images of some logos that we came up with.  let me know what
you think, I can post some GIF files once I get them from him.  Luke, you
probably have more time than he does to work on it to make it right.

If you guys like them maybe,

RespawnDespair, 2018-10-18:
Is there something missing from your message or is it just me?

RespawnDespair, 2018-10-18:
@htcohio i kinda like the 4th one [https://imgur.com/mMdn5CI](https://imgur.com/mMdn5CI). The others seem 'to big' for my taste. I'm by no means the one who decides on this, just my 2cts.

htcohio, 2018-10-18:
The gif file wouldn't have a black background...

I also have one where the EZ and ez wifibroadcast is capitalized but I kind of thought it looked cleaner all lower case.  What do you think?  

Either way, some kind of text image combined with Luke's logo would look amazing.

Yes21, 2018-10-18:
**@htcohio** I also prefer the 4th.

careyer, 2018-10-18:
The 4th one looks best yeah... however I think the Logo should rather be in icon size (simliar to dronebridge). This 1:1 ratio is way better for Avatars, Watermarks, Account images and so on.

htcohio, 2018-10-18:
Luke,

Since my images mostly include text, would you just want to duplicate this
concept on your end with a 1:1 ratio mentioned by @careyer?

bortek, 2018-10-18:
I actually like Luke's logo, its more graphical and representative, could be worked on a little bit by adding the tittle. I would like to see some of the @richardbmx , I know he is on it.

careyer, 2018-10-18:
@bortek : :+1:  The logos by @richardbmx look awesome. Very sleek and simple in design and very modern. For anybody who did not have a look yet: https://richardbmx.github.io/

htcohio, 2018-10-18:
> @bortek <https://github.com/bortek> : 👍 The logos by @richardbmx

Hey, I really like them all.

I was thinking about it and what if the logo that Luke threw together was
rotated to the left 45° so it was vertical and inserted into
ez-wifibroadcast as the "O" so that the curved Wi-Fi symbol is facing
upwards above the name

pilotnbr1, 2018-10-18:
@htcohio your 4th is my favorite as well. Curious to see what @richardbmx comes up with too

pilotnbr1, 2018-10-18:
@htcohio your 4th is my favorite as well. Curious to see what @richardbmx comes up with. As far as doing anything with the logo I threw out there, it is purely concept. Someone with better tools and an eye for that can rework it and add text. 

I think hd, ez, flight, radio, wifi, camera, communications are all ideas I hope someone can run with. I have to say I also like the abstract nature of dronebridges logo. Guess I am hoping something jumps out and grabs me!

richardbmx, 2018-10-19:
I would like to present my proposal based on all your ideas I have made this logo that expresses the camera and the Wi-Fi connection very well in addition to this it can be adapted to any background color and size without losing quality. If you want any change you should only ask for it.

[image]

bortek, 2018-10-19:
Ok let's give yo some more ideas. :) Since we are not only using this for camera but for a lot of other things like telemetry, OnScreenDisplay text, and some control functions. More accent should therefore be made more on the fact that this is an **easy** over the air solution rather than camera solution. The text should read "EZ-WifiBroadcast " with capital B. Can you think along this lines and . Some inspirations , look at these ones
https://www.logoground.com/uploads/2018137772018-03-054698924EZ.jpg
https://cdn2.vectorstock.com/i/1000x1000/98/01/ez-e-z-letter-logo-design-creative-icon-modern-vector-20689801.jpg
then text under or on the side "WifiBroadcast" plus some image to come up with.

bortek, 2018-10-19:
![](https://lh3.googleusercontent.com/WpckVDfK6UUd6vB4Yw4UCXMRx_umSvASUJYoatQERQLn49_V9sz_0xUc40Aoh4Hr4-M2SFKcATetg17tn9CjNvZnxyPKMokSOFruMDSkERyIOHGegY5AaVieumzTNxu86S_F1VRIVSrWYwK_-zLEuCFWx980ZUwLrN7hJ5qpDH_9jpSFkelHQ9APqO6bSJXMOtdiZcUTMWYubXp7qNr7bK3zbvi9e7QEAUT2xLLBY7su6OdQvKnro2fkgo81RbM3tsV8tTury2Wel6ugFUR5XKa-9-1FSe87pQmMe7nYSOOB9yGeq5JN2vAViVOctCbwgC2LqRk9NcSDKPhzwA-liwO_RTX_Fb0ipYedqfTmDyKyMTIsjqdjNuuQoy8ApLUsuIMXmAZ4Wt_afy8DS9DLt_QPlGtBIh90iwSd6KqdGTkxeQYa3DoknpgnHhc_cbufI3ObkM7wq3cGFYf87cEUfHSY_NJGckMDAZcReSUFHtQP_E8G_ig-xdIVqtO2pzxTOyodMNhj5LWdBL0vC-1Xe9aDBR9Hm2h8XxyV2VK-JEDYp6UYQMLiptv2dbvasFvWOeCg1DHeA7fvvxeW_sxCosszPNiGMG3qRp5DU5PYxGkJ39hEx6rXGM9FyJt0lB8MMlVuNe7O_QR60GX3Vs1godfqItJKoSMbxvmnrviVPWdvxdQe5AnuMAZEIw=w324-h256-no)

careyer, 2018-10-19:
@richardbmx : I like your first draft a LOT!!! Very classy... However I can join the word of @bortek:

> Ok let's give yo some more ideas. :) Since we are not only using this for camera but for a lot of other >things like telemetry, OnScreenDisplay text, and some control functions. More accent should [...]

Please, can you take a second try and try to incorporate those thouhgts of bortek? I will try to cut the essence in a shot formular:  **HF-Video + Telemetry + RC-Control ==> EASY!**
Maybe this helps! @richardbmx - thank you very very much for your effort!

RespawnDespair, 2018-10-19:
@richardbmx The level of competence is clear from your designs, i do feel, as the others do, that it does not fully reflect the project. I would confuse it with the camera app on my phone 😄 
If you can indeed emphasize the 'HD' and 'EZ' in the logo as well as the 'WiFi' (which you have now). I don't think it's possible to show all the project can do in one logo, but the 'EZ' and 'HD' and 'WiFi' are at the core of everything it does...
It's hard to please everyone, and let me repeat i absolutely love your style!

bortek, 2018-10-19:
I added 9 images in total but only one showed here, doh :( Now trying again :)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

pilotnbr1, 2018-10-19:
My vote so far from most to least favorite 
[image]
[image]
[image]
[image]

@richardbmx I like you designs but I think the camera aspect is too prominent and not enough “radio” “transmission” element... Or go more abstract like in @bortek designs. Thanks for the effort! Very nice!

careyer, 2018-10-19:
I hope @richardbmx takes another try... From what I have seen on his facebook page he beats us to hell with his skills and his experience in design. I think we must go through different iterations first to make a decision. How is in for that?

pilotnbr1, 2018-10-19:
I know I'm muddying the waters.. but personally I wish it was just WifiBroadcast and dump the EZ. WBC for short.

bortek, 2018-10-19:
Thats the name that befis original project was called and that now too broad and points to quite broad range of the search results. We will need to come up with the new name in the future if we want to.

htcohio, 2018-10-19:
@Bortek,

I agree, and a new name that incorporates a phrase "By: EZ-WifiBroadcast" or something could still maintain the legacy of the project. 

My thoughts on a name change:

1.)  I heard comments about this project a few times in the FPV.Blue forums a year prior to picking it up but, the general consensus for long range wireless HD has been "Wifi can never be stable or reliable". It wasn't until fpv.blue stalled out that I started researching further and started to understand the concept of packet injection and monitor mode which helped me understand what made this system unique.

2.) I like to think that I bring that certain point of view as a total beginner who didn't know the first thing about GitHub, what an "image", or Raspberry Pi was for that matter (no making fun of me!!!) lol... I remember being confused at first about what web pages, instructions or projects were correct because web search info for wifibroadcast went back to 2015/2016 and those instructions were more "code/linux" based and outdated. Eventually I found the right page and rcgroups which helped a lot but until that point  I never really referred to forums for information. 

As you said, it "points to a broad range of search results" and that part has only become more amplified as of today.

Given the new life of the project and proven success, I believe a new, more eye-catching name would only spark new interest into the project while it is still relatively small in comparison to others.  Furthermore, this dedicated community of awesome people will work to spread the info and point newcomers in the right direction.  

It would be easier to manage a name-change sooner than later when it becomes more popular (which I know it will).  I do think the benefits outweigh the (rebranding) drawbacks but, it is certainly worth at least talking about and it could be fun working together to come up with something new :)

@RespawnDespair suggested the name "Open.HD" a little while back which I think perfectly represents what the project is.

Some other ideas for consideration and they could be combined with many other titles/ideas:
-OpenHD
-EZ-HD 
-Pi-HD
-Bortek.HD
-AthRalink.HD (name merged from the 2 popular Atheros and RaLink wifi chipsets).
-ClearBroadcast
-Befinitivs-HD

Anyways, there's my 2 cents :)

pilotnbr1, 2018-10-19:
I think at this point @bortek is correct that we should wait on any name change. As of now there is renewed interest and progress on this wiki but we haven’t released any new software. I think we will with everyone’s hard work, but software development is fragile with only a handful of devs actually doing the heavy lifting. I say we get at least one good new release under our belt and then contemplate a name change. I get why it would make sense earlier than later but some things need to be incremental, be accepted, and take time... Sorry I brought up a name change :/

careyer, 2018-10-19:
+1 keep the name

richardbmx, 2018-10-20:
I am working on this I am developing different ideas when I have it ready I share greetings.

richardbmx, 2018-10-21:
I present this second idea that is formed by the union of EZ and WIFI which in my opinion is the main idea of the project, it is a simple and modern minimalist image that helps to be remembered easily and identifies the project. It is important to know that not all the logos express all the functions of the application. The important thing is to make the logo known so that the people to see it know that it belongs to the project.

I will continue working on another proposal when I have it ready, I would like to present it to you, however I would like to know your impression this new proposal

[image]

richardbmx, 2018-10-21:
Third proposal that came to my mind very fast and that I think looks very nice to look at and expresses very well the WIFI connection of the application

[image]

careyer, 2018-10-21:
> Ohhhh i love that second set!

Yeah!!! The second Set looks awesome! I really like it. It is very reduced to the very essential and can be easily remembered. Kudos!!!

pilotnbr1, 2018-10-21:
@richardbmx very nice!

My favorite is
[image]
[image]

bortek, 2018-10-21:
@richardbmx Sweet. I like this one

[image]

careyer, 2018-10-21:
Yes this is also my favrorite one... It's just ingenious how the "e" and the "z" blend together!

richardbmx, 2018-10-21:
Perfect I'm glad you like it, you just have to decide which one you want as definitive to make a Pull requests  and a google folder with all the necessary files for the use of the logo. ;-)

Yes21, 2018-10-21:
> Yes this is also my favrorite one... It's just ingenious how the "e" and the "z" blend together!

I also prefer this one.
But I don't like the dark color of the "Z". I would prefer orange (like in my avatar) ... ?

richardbmx, 2018-10-21:
@pilotnbr1  All my designs are free to use, the contributions are made with him in order to grow my design portfolio and contribute in some way to the open source community, with mention to me in the readme as creator of the logo for my is enough

richardbmx, 2018-10-21:
A variation with orange @Yes21 

[image]

Yes21, 2018-10-21:
**@richardbmx**  Cool ! I love it much better like this.
I prefer the left one.
What do the others think ?

pilotnbr1, 2018-10-21:
I like the left design. The color change is neutral to me, so either color works for me

richardbmx, 2018-10-21:
The title text should be the same color as the one in

htcohio, 2018-10-21:
+1 Left, +1 any color

bortek, 2018-10-21:
Looks good in orange too :) I also vote for the left one. Now about the text. I think it is supposed to be the same orange color so it blends or it could be black too. But what about the text itself? Since the Logo already says EZ then the rest of the text could be WifiBroadcast. Otherwise it reads like EZ EZ-WifiBroadcast. What do you think?

pilotnbr1, 2018-10-21:
@bortek +1.   So it should be LOGO - WifiBroadcast ?

bortek, 2018-10-21:
Asked a few people around here how they read/interpret and they think the text should read "LOGO EZ-WifiBroadcats" because people not familiar with project  treats the logo (EZ blend ) NOT as two letters but more like an emblem. So perhaps it should be "LOGO EZ-WifiBroadcats"

> The title text should be the same color as the one in

@richardbmx We can of course mention you on first page e.g. under the license part. Can we get the source files (SVG or photoshop or whatever editor you use) in case if we need to edit?

pilotnbr1, 2018-10-21:
@bortek glad you did some “testing”! Yes makes sense. Leave it “logo EZ-WifiBroadcast

careyer, 2018-10-21:
@richardbmx : ...+1 what @bortek  said and maybe you can follow allong the project and maybe contribute buttons or graphics elemenety whenever needed? It is just a nice thing to have a consitant look all over the website/wiki

richardbmx, 2018-10-21:
I was referring to this. the text can go the same color orange or blue or dark gray in any of the three ways it looks great

[image]

richardbmx, 2018-10-21:
Perfect I hope you confirm the definitive one to generate all the files

Yes21, 2018-10-21:
**@richardbmx**  Ok for me.
Waiting for the others opinions ...

careyer, 2018-10-22:
Yes for Yes21 as well

bortek, 2018-10-22:
All 3 looks nice to me but for the sake of choice i take too with blue text. Justin combination of blue-orange is the compatible one for human eye/brain that's why many choose it. 😄

https://qph.fs.quoracdn.net/main-qimg-0f59f02f66a465082ef7511320a62be9

richardbmx, 2018-10-22:
Perfect I will generate all the necessary files for the use of the logo and I will submit an pull request

pilotnbr1, 2018-10-22:
I defer to you guys, all of these choices from Richard look awesome!

Yes21, 2018-10-22:
**@richardbmx** Many thanks to You.

I will use the 115.svg right now for the splashscreen of the next 1.6 ez-wbc release.

richardbmx, 2018-10-22:
perfect I made one pull requests loading the logo and placing it in the Readme

pilotnbr1, 2018-10-23:
Very Nice!

careyer, 2018-10-23:
SWEET :-)

richardbmx, 2018-10-29:
This looks really cool I like how it looks on an excellent black background, I like to know that you liked my work a great pleasure for me to work for you

careyer, 2018-10-29:
Logos online -- will close this issue for now! Thank you for everyone to contribute to this.

closed 2018-10-29 by careyer