rodizio1/EZ-WifiBroadcast #167: What should I modify if I want to send custom bytes instead of telemetry and rc?

09Ria09, 2018-10-18:
Sorry if this is not the place to ask. I already wrote code for both the GroundPC and the AirPi and the only thing left to do is the video connection and the data connection. I found out about this project 2 weeks ago, but I'm quite bad at reading other people's code. So if you could help me by giving a few hints, that would be nice

pilotnbr1, 2018-10-19:
@09Ria09 
I’ll take a stab at this.. be gentle as I have not looked at this code in some time and it is now much more complicated than what I played with.

Check out 
https://github.com/bortek/EZ-WifiBroadcast/blob/develop/wifibroadcast-scripts/profile

Most of the magic happens there. Specifically the tx function and rx function. It appears that on the airpi side the pi is simply echoing what is seen on the flight controller serial port with the “cat” call with the osd function. Hope that is helpful and gets you going!

htcohio, 2018-10-19:
Hello,

I believe user1321 already did this couple months ago.  It is not anywhere
near being integrated with the image Builder but I'm sure he would share
anyting you need with you.

Here's a couple of the videos.

https://youtu.be/bwGPdL-97QA

https://youtu.be/sevEihOZncQ

user1321, 2018-10-19:
Rx and tx program use  streams as input and output, if you wish to use udp packets you may look at this project

https://github.com/svpcom/wifibroadcast

also in case of tx, rx you dont need to use instead, you may use in same time. 

for example : rx -p 32 ... 

where -p  is a some kind of channel .

you may start /rx program from command prompt and take a look at output

09Ria09, 2018-10-22:
@pilotnbr1 , @htcohio  , @user1321  and @dposea  Thank you all for helping me!
2 more questions: I keep seeing this: `nice /root/wifibroadcast/rx_rc_telemetry -p 3 -o 0 -b $FC_RC_BAUDRATE -s $FC_RC_SERIALPORT -r $RC_PROTOCOL $NICS &` in `profile` but there is no `rx_rc_telemetry` file in the source code, there is only on the raspberry pi running the image v1.6RC6 . And also where is the `profile` in the image?

@user1321 Do you know how much faster is the project using udp packets? It says on the page it is faster than EZ-Wifibroadcast but it doesn't say by how much. I will start looking over https://github.com/svpcom/wifibroadcast code right now.

pilotnbr1, 2018-10-22:
@09Ria09 I am traveling right now and as said before it’s been a long time (years) since I have looked at the code.

I believe .profile is in root and it might be hidden. So make sure you are able to view all files when exploring the image.

The github code is being redone, some is outdated and not complete. I suggest you just use your image as source.

I am sure some of these other users here will have better more up to date info.

Which way are you trying to send data, ground to air or air to ground? Or bidirectional?

09Ria09, 2018-10-22:
@pilotnbr1 Yeah, you're right, `profile` is under root and it's hidden. I'm trying to send bidirectional data, using the Multiwii Serial Protocol.

09Ria09, 2018-10-22:
@pilotnbr1 Well, I mostly started working on this project to learn, so I kinda reinvented the wheel. My project is using Python and JavaScript(and HTML and CSS).
The python code on the ground launches a website which is the interface and communicates with the air pi. The interface has 2 tabs: WayPoints Tab and Video + Controls Tab. So far I've tested my code using 2 HC-12 modules (no video, only data). And now I want to use only WIFI and get rid of the 2 HC-12.
The python code on the air translates data into `MSP `and vice versa. I use `MSP v1` because I couldn't get the `crc_dvb_s2 checksum` to work. But it would be easy to change it to `MSP v2` if I know how to implement the `crc_dvb_s2 checksum`.
@user1321 Proposed this: https://github.com/svpcom/wifibroadcast and it seems way easier to send custom data using it.
I also noticed this function `mspdownlinktx_function` in `profile`, but I forgot to mention that on my drone there is a servo connected to the Raspberry Pi which is controlled by sending different data, but I guess I could assign that value to an unused `RC channel`, and the Pi on the air would check that value and act acordingly.
Currently I'm still trying to understand the EZ-Wifibroadcast's code, but I think I'm getting closer to a solution.