rodizio1/EZ-WifiBroadcast #168: Verify Status of MSP

pilotnbr1, 2018-10-22:
It is unknown if MSP is a working RC protocol. I will take a look at this one next week.

zipray, 2018-10-23:
Yes it is

careyer, 2018-10-23:
@zipray : Can you shed some light on this maybe? What is the current status of MSP support in the latest 1.6rc6 build? Is it supporting MSP v1.0 or MSP v2.0 and - most importantly - is it enabled at all? The release notes mention nothing about it and as both of us @pilotnbr1 and me can remember there was a discussion long ago that there were version problems with v1 and iNav (only supporting v2). That ultimately resulted in the need for the MSP implementation be be overhauled and temporarily I remember that it was deactivated completely?

zipray, 2018-10-23:
@careyer @pilotnbr1 
With RC=MSP enabled in the WBC, control can be achieved by connecting the air pi tx to the RX of the flight control RC serial port.But like mavlink, the air pi 's serial interface, rx,tx, is occupied at the same time.Even if I connect the airborne rx to the airborne telemetry serial port tx, and enable the LTM telemetry protocol, the information cannot be obtained in the ground OSD.In the case that both the FC and ari PI serial ports are defined as MSP's and cross connection, I do not know if any MSP data is returned to ground pi, as the OSD does not support MSP data display.

pilotnbr1, 2018-10-23:
@zipray thanks for that! Very much appreciated! How much have you used the msp rc? Has it been reliable?

@careyer
We dont truly know what is in the ez code. I am going to investigate this further from the code side to confirm what version of msp is running. The assumption is that ez is v1 and inav msp v2 is backwards compatible.

careyer, 2018-10-23:
@pilotnbr1 : Yes please dig into the code... but I believe this says it all:
https://www.rcgroups.com/forums/showpost.php?p=40085007&postcount=5497

zipray, 2018-10-23:
@pilotnbr1 @careyer Thank you very much for your reply.
I've heard horror stories about MSP that I didn't encounter on inav.
MSP connection cannot be obtained by connecting PC and ground PI serial port via usb-ttl module, including uplink MSP navigation point or adjustment parameter and downlink FC state.I think WBC only implements MSP RC output of air PI, because the MSP v2 of inav is compatible with v1, MSP RC is available at present.

rodizio1, 2018-11-05:
> @pilotnbr1 : Yes please dig into the code... but I believe this says it all:
> https://www.rcgroups.com/forums/showpost.php?p=40085007&postcount=5497

No, this doesn't say it all, also read my reply to Target0815's posting:
https://www.rcgroups.com/forums/showpost.php?p=40085099&postcount=5499

(The change that made it incompatible wasn't the introduction of MSP v2, it must've been something else, as it stopped working somewhere around Cleanflight 1.13 or 1.14, well before MSP v2 was introduced.)

But, ofcourse, _now_ it probably makes sense to implement MSP V2 as Cleanflight/Inav/Betaflight use that.

zipray, 2018-11-05:
@rodizio1 It's nice to have you back and continue to lead us through the WBC's ""hogwarts" .

pilotnbr1, 2018-11-05:
Well based on @rodizio1 comments, (thanks for shedding more light on it) i think msp needs an update (especially if it’s easy to implement). I suggest that be a new issue or enhancement, but I think we can close this issue as far as the question goes.

closed 2018-11-06 by bortek