rodizio1/EZ-WifiBroadcast #170: RC only working with Atheros AR9271

careyer, 2018-10-23:
> That might verry well be the case. I am not sure though. All I know is that the RaLink cards are not capable to be used for RC. Since RC is just part of the Uplink-telemetry stream.... yeah well possible.

If it works only with old atheros chipset, the scope of this project is becoming very reduced !

I could see the uplink telemetry work only one time : with the last "russian image", and my rtl8812/14au boards ...

_Originally posted by @Yes21 in https://github.com/bortek/EZ-WifiBroadcast/issues/165#issuecomment-431956693_

careyer, 2018-10-23:
@Yes21 : Please find above the answer about the topic that we were talking about just yesterday....

Yes21, 2018-10-23:
**@careyer** : Thanks.
Do you think that uplink telemetry is linked with RC capabilities ?
In both cases the groundPi should send orders to the airPi ...

careyer, 2018-10-23:
@Yes21 : I cannot really tell. RC-control is part of the MAVlink protocol - that is for sure. However the MAVlink protocol is generated at the AirPi - so to say at the last step in the transmission pipeline. So one needs to dig into the code and see if the RC signals (the raw data that is sent via the radio link) are transmitted exactly the same way as the uplink telemetry payload. If this is the case we just found the issue you are experiencing. Sadly I am not a coder.

zipray, 2018-10-24:
@Yes21 We met in "Russian image".
Attempts to pass the 5.8g uplink telemetry and RC were possible only in the case of a single receiving WiFi adapter.
I have three WiFi adapters similar to AWUS1900, and when one of them is on air PI and two are on ground PI, telemetry and rc will disconnect shortly after startup.But with both air PI and ground PI using one WiFi adapter, everything is fine.

pilotnbr1, 2018-10-24:
@zipray thanks for the info! Very helpful while we backwards document this project

pilotnbr1, 2018-10-24:
Another part of the rc mystery is the code in the arduino folder msp2ppm. Also a post on rcgroups referencing msp to ppm

https://www.rcgroups.com/forums/showthread.php?2664393-EZ-WifiBroadcast-cheap-digital-HD-transmission-made-easy%21/page120

I am guessing the groundpi sends rc as msp and the airpi converts to ppm outputs. But anemostec references pin 10 which is an RX pin so dont know... Just thought I would throw some more gas on this RC dumpster fire lol

Yes21, 2018-10-24:
**@zipray** 

> @Yes21 We met in "Russian image".
Attempts to pass the 5.8g uplink telemetry and RC were possible only in the case of a single receiving WiFi adapter.
I have three WiFi adapters similar to AWUS1900, and when one of them is on air PI and two are on ground PI, telemetry and rc will disconnect shortly after startup.But with both air PI and ground PI using one WiFi adapter, everything is fine.

Thanks. I can confirm that uplink telemetry did work (with last russian image) for me with only one wifi adapter on each side.

So that confirms that the 8812/14au adapters are RC and up telemetry compatible ...

careyer, 2018-10-24:
That's all good news. At least there is hope that we can extend the RC capability beyond a specific chipset. It is really crucial to gather all the information and understand what exactly is happening and which parts of the code are still active and which ones are legacy code. I know that Rodizio was struggling with an own RC implementation which had bad latency and stability issues in the first place when dino.de stepped in and donated some code. Since the new code worked without problems I think this was adapted.  And yes it is very likely that we find some abandoned code fragmenets from Rodizios first tries. Thanks for all trying to shed some light on this against all odds of the totally neglected documentation.

Yes21, 2018-11-05:
I can now affirm that up telemetry is working well with rtl8812/14au, with the latest building script image.
So the AWUS036ACH and AWUS1900 should be RC able, but didn't tried it yet ...

zipray, 2018-11-05:
>           I can now affirm that up telemetry is working well with rtl8812/14au, with the latest building script image.
> So the AWUS036ACH and AWUS1900 should be RC able, but didn't tried it yet ...

Yes, I have tested RC and it works.However, bandwidth usage is not normally increased.

careyer, 2018-11-05:
@zipray : no you should not notice any impact on the downlink performance. Did you use multiple Wifi Adapters at the GroundPi or just one?

Why am I asking? User FlingW on RCgroups had a strange problem with servos moving rather sticky/erratic and not smooth on his ArduPLANE. Nobody else reported this behavior before. In the progress of debugging we found out that he was using only a single Wifi Adapter at his GroundPi and not multiple sticks which is common good practice. Obviously the only Wifi Adapter at the groundpi was rather busy with receiving downstream packets and could not send the uplink packets (which includes RC) with the required datarate. Once he added a 2nd Wifi adapter everything runs buttery smooth.
I think we should investigate in this behavior .
Maybe this has something to do with your observation?

rodizio1, 2018-11-05:
R/C has been designed to work with Atheros only, if it works with other chipsets, that's just coincidence.

>Obviously the only Wifi Adapter at the groundpi was >rather busy with receiving downstream packets and >could not send the uplink packets (which includes >RC) with the required datarate. 

You conclusion/explanation is wrong. Using a 2nd adapter shouldn't make a difference. After all, both adapters receive the same packets from the TX and thus, both are similarly busy. There must be a different explanation.

careyer, 2018-11-05:
You are right Nico.... It was just an educated guess. However adding a 2nd adapter immediately remediated the issue.. There is a algorithm involved which decides which stick sends the uplink data depending on the RSSI value, is it? Maybe to look there?

Yes21, 2018-11-05:
@rodizio1 So can you explain us what is the link between RC capability and telemetry uplink capability ?

rodizio1, 2018-11-06:
>This has long time been a rumour and nobody could answer why

Rumor? It clearly says "Atheros only" in the feature list:
https://github.com/bortek/EZ-WifiBroadcast/wiki/General-~-Features
>Low-latency/high update-rate RC over wifibroadcast via Joystick (Atheros only)

>So can you explain us what is the relation between RC capability and telemetry uplink capability 

There is no relation, telemetry uplink and R/C are handled by different programs/code in the image. To make R/C work with other adapters, somebody needs to actually dig into the drivers/kernel/radiotap headers of Ralink and Realtek adapters.

zipray, 2018-11-06:
@rodizio1 Therefore, I  understood that if RC=mavlink is enabled, whether uplink=mavlink or disable, RC andTelemetry should not be performed on the same serial port.Isn't it?

careyer, 2018-11-06:
> Rumor? It clearly says "Atheros only" in the feature list:
> https://github.com/bortek/EZ-WifiBroadcast/wiki/General-~-Features

Yes this  is true. However people by accident found out that RC also works (or kinda works) with non-Atheros cards. A users on RCgroups even reported that a mixture of Atheros (I believe Air) and RaLink (Ground) adapters will do. This is what I was referring to as "rumor" ;-)
So, indeed,  at the end of the day it reads "(Atheros only)" but people see that it works with their other setups as well and that is what confuses them. That is exactly the same thing that we are now observing with rtl8812/14au as well.

I think we need to have more insight into what prerequisites make RC work or not work because test results show that "Atheros only" does not cut it anymore.

rodizio1, 2018-11-06:
@zipray R/C and telemetry can be sent over the same serialport, there is nothing wrong with that. 
See here for possible options: https://www.rcgroups.com/forums/showpost.php?p=39575125&postcount=4174

@careyer 
>Yes this is true. However people by accident found out that RC also works (or kinda works) with non-Atheros cards. A users on RCgroups even reported that a mixture of Atheros (I believe Air) and RaLink (Ground) adapters will do. This is what I was referring to as "rumor" ;-)
So, indeed, at the end of the day it reads "(Atheros only)" but people see that it works with their other setups as well and that is what confuses them. That is exactly the same thing that we are now observing with rtl8812/14au as well.

I know for sure that it doesn't work with Ralink at the ground Pi. 

Atheros at the ground and Ralink in the air might work (I have never tested that though), but due to the Ralink adapter in the air not sending CTS protection frames, packetloss will be higher. 

Mixed Ralink/Atheros on the ground will lead to packets only being sent by the Atheros adapters (and currently, I'm not sure what happens if the Ralink adapter sees a higher signal strength, it could be then that the Atheros never gets used).

About RTL8812/14 I cannot really comment, as that chipset never was officially supported with EZ-Wifibroadcast.

>Does (1) mean, that when using RC=mavlink this will force CTS to be enabled? I think we all are missing bits and pieces of information. It would be great to shed some light on this.

No, it isn't forced automatically, it has to be set manually, see here, in my original documentation:
https://github.com/bortek/EZ-WifiBroadcast/wiki/How-to's#setting-up-rc-over-wifibroadcast-with-a-joystick

zipray, 2018-11-06:
[image]
[image]
[image]

careyer, 2018-11-06:
Alright! I did some extensive testing this evening and these are the results:

1.6RC6
FREQ=2472
DATARATE=4
TELEMETRY_TRANSMISSION=wbc
TELEMETRY_UPLINK=mavlink
RC=mavlink
CTS_PROTECTION=auto
WIFI_HOTSPOT=Y  (This is what I connect through)

**Setup1:**
- AirPi = 1x ALFA AWUS036NHA (Atheros AR9271)
- GndPi = 1x TPLink TL-WN722N V1.10 (Atheros AR9271)
- Video = OK
- Telemetry = OK
- RC = OK
- Mission Planner: All connects buttery smooth

**Setup2:** 
- AirPi = 1x ALFA AWUS036NHA (Atheros AR9271)
- GndPi = 1x CSL 300Mbit (Ralink RT5572)
- Video = OK
- Telemetry = OK
- RC = OK  **<-- This should not work but does**
- Mission Planner: HUD information updates, "Getting params" does not even start

**Setup3:** 
- AirPi = 1x CSL 300Mbit (Ralink RT5572)
- GndPi = 1x CSL 300Mbit (Ralink RT5572)
- Video = OK
- Telemetry = OK
- RC = FAIL 
- Mission Planner: HUD information updates, "Getting params" starts but gets stuck on first parameter 

All tests repeated several times. Results were consistent.

careyer, 2018-11-09:
Thank you very much for pointing this out. This makes total sense and is defintely a thing that we should have an eye on in one of the upcomming releases as Atheros cards are increasingly now hard to come by.

zipray, 2018-11-09:
@careyer Is your mavlink test done in routing mode or node mode?When I use both RC and teletemtry at the same time, they are quite different.

careyer, 2018-11-09:
@zipray : Honestly I have no idea if it uses mavlink-routerd oder cmavnode mode. What I can tell is that I alsways have RC and telemetry active by default so my tests always reflect the result of RC+TELEM. Valuable information that the MavLink communication with RC+TELEM and TELEM only look significanntly different! Thanks for sharing

Yes21, 2018-11-14:
I can now confirm that RC is not working with RTL8812/14AU boards (with latest 2.0beta).

Yes21, 2018-11-14:
@zipray Sorry, I don't understand your question.

Yes21, 2018-11-19:
@rodizio1 
I have many questions about wifi adapters RC compatibility :

- should only the wifi groundPi adapter be "compatible" ?
- are the only prerequisites CTS protection and RTS frames injection  ?
( what means CTS and RTS ?)
- how to know if a wifi card is CTS/RTS compatible ?
- could you explain exactly what is needed to be done for making work RC with a particular wifi board ?

I'm asking all these questions, thinking of RTL8812/14 boards ...
I'm pretty sure they are RC able, and I hope you will point us into the right way to make them work for RC.

Many thanks to you.

closed 2018-11-23 by careyer

Yes21, 2018-11-26:
This issue has been closed "accidentally"  by @careyer last black friday !
@bortek Could you please reopen it ?

Howie9600, 2019-07-30:
Has this been sorted yet?  I really need RC working and having massive issues

pilotnbr1, 2019-07-30:
No..  no development taking place here. Try OpenHD