rodizio1/EZ-WifiBroadcast #171: Why is this project not being used commercially?

myeabhishek, 2018-10-25:
Hello Team,

On going through this repository, I realised this is great work that you guys have accomplished. Could you please help me understand why is this not being used commercially? Are there any known issues with this project? Or is there some licensing issue?

RespawnDespair, 2018-10-25:
Most, of not all open-source projects have a license prohibiting commercial gain directly from the project.
E.g. You can't ask money for the project itself or it's direct output.
The same also goes for derivatives, so you can't just fork the project, change the name and then charge people money.

Now, this only counts for the project ITSELF. You can make custom hardware or bundles of existing hardware, sell that with the software on it, as long as you EXPLICITLY do not charge people for the software.

So i hope to see some nice hardware bundles or even custom hardware solutions running EZWFB in the future. I know for a fact some of the contributors on the project are planning exactly that. It would be a tremendous boost to the project as more people would start using it.

Now as to issues, yes, there are still a lot of issues, as evidenced by the videos and the comments the project works well in most situations. There are however a lot of custom images, each with it's own specific tweaks. There is now a project underway where we are merging all these images into one solution. 

Once this new 'master' image is released we should have a stable version of EZWFB which includes a lot of optimizations and extensions developed by the community over the last years.

Such an image could (and hopefully will) be used by commercial entities offering it on their (custom) hardware.

Sorry for the long text, but i hope it clears this issue.

myeabhishek, 2018-10-25:
Thanks RD for the explanation. You mentioned there are still lots of issues still with the project. Is there a list that I could refer to see these issues?

RespawnDespair, 2018-10-25:
https://github.com/RespawnDespair/wifibroadcast-image-builder/projects/1

This is ongoing, so bugs are added, but the majority is fixed or known.
Pi3B+ support and 8812AU support are the major issues.

Could you share some background as to your interest in the project? Do you want to commercialize it in some way? Pure interest from me.

careyer, 2018-10-25:
@myeabhishek : There is not a comprehensive list of all issues but major ones are:
- No support for up-to-date SOCs (Pi3B+)
- No support for modern and readily available Wifi chipsets
- Only legacy support for old versions of some protocols (e.g. MavLink and MSP)
- no final and proper implementation of the regular RC protocols (e.g. they do not support all channels)
- poor to no documentation for parts of the software

We are woking on all that right now.

myeabhishek, 2018-10-25:
I am experimenting on opencv. I came across this project for a different requirement. I asked this out of pure curiosity.

closed 2018-10-25 by myeabhishek